lyusupov/SoftRF #1: garbage in GPS data

biturbo, 2017-02-21:
Hi I was just checking your code and I was not able to get a clean NMEA data out on TX (GPIO1), do you have any idea why, NMEA sentence are correct on D3 (GPIO0), I reduced the baud to 9600 on both GPIOs

here an example of data with garbage:

$GPGSV,3,2,12,02,30,309,,03,26,110,45,40,19,1¦ i’bŠÊb‚ÊŽIš²bšŠRºÊj
$GPGGA,081102.000,4735.0051,N,00NM&é’‚¢b*±1,08,1.25,430.4,M,48.0,M,,*68 $GPRMC,081118.000,A,4735.0051,N,00852.7203,EKr‚‚b‚r‚‚b’Š‚’Šîbbb
$GPGSV,æIbŠ’b‚’bš‚bš‚ÊbŽ	šb’²bŠŠ‚b¢ªb¢‚Ž)ÊbŠ’ªbš’bŠÊb‚ÊŽIš²bš‚RºÂj
$GPVTG,$PFLAU,1,1,2,1,3,-102,2,44,4*4A

lyusupov, 2017-02-21:
Due to hardware restrictions I had to use SoftwareSerial to pick data from ublox.
I suspect that SwSerial is affected by external activity such as Wi-Fi traffic.
My current choice is to limit NMEA sentences to a minimum - GxRMC and GxGGA only.
I've used "U-center" to get rid of other messages (as well as to set up "Aircraft-2g" kinematic model).

I estimate that loss of NMEA input to about not more 10%.  Which is acceptable for this moment.

biturbo, 2017-02-26:
Would it not be better to use Serial1 (GPIO2/D4) as the TX for BTRX and the Serial (GPIO3/RX) for GPSTX so you would not use SoftwareSerial, just an idea.

lyusupov, 2017-02-26:
This is the ESP8266  GPIO pins allocation map that I've taken into account:

[image]

With NodeMCU I am unable to manage GPIO3 for hardware serial input because this pin is connected to CP2102 through  470 Ohm resistor. See this schematic for details: https://github.com/nodemcu/nodemcu-devkit-v1.0/blob/master/NODEMCU_DEVKIT_V1.0.PDF

I saw no information on how to put CP2102 into off mode or any other mode which would leave this Tx line into 'free' state. When t is not free - 470 Ohm is too heavy load for GPS module to override.
 
If you suffering a lot from this "garbage" you may consider to solder yourself ESP-12 onto a PCB which would have same form factor and pinout as NodeMCU.

I earlier decided to leave this issue "AS IS" as a known compromise.

lyusupov, 2017-02-28:
An illustration that reduction of NMEA input to a minimum does help.
Ublox "factory" settings:
```
$GPGLL,,,,,155832.00,V,N*42`
$GPRMC,15583&   ▒b
                  ▒5600.54310,N,822.22946,E,155833.00,A,A*60
,10,215,20,17,47,063,*7A
$GPGSV,5,3,17,19,64,087,,22,05,025,,24,68,240,29,25,07,271,32*7E
$GPGSV,5,4,17,28,06,087,,32,19,323,,37,24,200,27,39,25,196,27*75
$GPGSV,5,5,17,42,,,▒▒▒5
                       $GPGLL,,,,,155834.00,V,N*44
$GPRMC,155835.00,V,,,,,,,280217,,,N*7C
$GPVTG,,,,,,,,,N*30
$GPGGA,155835.00,,,,,0,03,5.63,,,,,,*5A
$GPGSA,A,1,15,25,12,,,,,,,,,,5.72,5.63,1.00*01
$GPGSV,5,1,17,01,05,018,,02,7C
$GPGLL,,,,,155835.00,V,N*45
0,215,20,17,47,063,*74
$GPGSV,5,3,17,19,64,087,,22,05,025,,24,68,240,31,25,07,271,33*76
$GPGSV,5,4,17,28,06,087,,32,19,323,,37,24,200,27,39,25,196,27*75
$GPGSV,5,5,17,42,,,27*7C
$GPGLL,,,,,155836.00,V,N*46

$GPGLL,5600.54121,N,03822.22841,E,155837.00)▒A*62
$GPRMC,155838.00,A,5600.54078,N,03822.22691,E,1.194,,280217,,,A*79
$GPVTG,,T,,M,1.194,N,2.211,K,A*2E
$GPGGA,155838.00,5600.54078,N,03822.22691,E,1,03,5.62,187.5,M,12.5,M,,*59
$GPGSA,A,2,15,25,12,,,,,,,,,,5.71,5$PFLAU,1,1,2,1,1,-64,2,251,778*43
$PFLAA,1,341,-699,251,2,FCA567,,,,,1*2F
3,27,37,24,200,29,39,25,196,29*79
$GPGLL,5600.54078,N,03822.22691,E,155838.00,A,A*63
41,27,15,10,215,21*7C
$GPGSV,5,3,17,17,47,063,16,19,64,087,,22,05,025,18,24,68,240,32*7F
$GPGSV,5,4,17,25,07,271,33,28,06,087,,32,19,323,,37,24,200,28*78
$GPGSV,5,5,17,39,25,196,28*46
$GPGLL,,,,,155839.00,V,N*49
$PFLAU,1,1,2,1,1,-66,2,251,831*43
$PFLAA,1,338,-759,251,2,FCA567,,,,,1*2C
,26,15,10,215,20*7C
$GPGSV,5,3,17,17,47,063,12,19,64,087,,22,05,025,,24,68,240,31*71
$G▒▒eŪb▒b▒▒b▒▒b▒▒b▒▒▒b▒▒b▒▒b▒▒b▒ºbb▒▒b▒▒b▒▒▒bb▒▒b▒▒b▒▒▒b▒▒R▒▒j
                                                              $GPGSV,5,5,17,39,25,196,27*49
$GPGLL,,,,,155840.00,V,N*47
$GPRMC,155841.00,V,,,,,,,280217,,,N*7F
$GPVTG,,,,,,,,,N*30
$GPGGA,155841.00,,,,,0,00,99.99,,,,,,*6A
$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30
$GPGSV,4,1,16,01,05,018,,02,09,160,,03,02,045,22,04,,,,,,,,280217,,,N*7C
!▒u025,,24,68,240,31*72
$GPGSV,4,4,16,25,07,271,32,28,06,087,,32,19,323,,39,25,196,27*75
$GPGLL,,,,,155842.00,V,N*45
,155843.00,A,A*66
,155844.00,A,A*60
2,02,09,160,,03,02,045,,04,,,25*49
$GPGSV,4,2,15,06,28,127,,12,47,270,40,14,05,341,25,15,10,215,20*72
$GPGSV,4,3,15,17,47,063,,19,64,087,5600.53860,N,03822.22878,E,1,04,11.48,189.6,M,12.5,M,,*68
$G▒▒▒,07,271,31,28,06,087,18,32,19,323,27*4F
$GPGLL,5600.53860,N,03822.22878,E,155846.00,A,A*65
00,E,155847.00,A,A*6D
``` 
GxGSV, GxVTG, GxGLL GxGSA are disabled. ( TinyGPS+ ignores them anyway ):

```
$GPRMC,174308.00,A,5600.56151,N,03822.26892,E,1.357,,280217,,,A*7E
$GPGGA,174308.00,5600.56151,N,03822.26892,E,1,04,4.73,151.0,M,12.5,M,,*5B
$PFLAU,1,1,2,1,1,-61,2,287,1074*77
$PFLAA,1,520,-939,287,2,FCA567,,,,,1*20
$GPRMC,174309.00,A,5600.56125,N,03822.26864,E,1.804,,280217,,,A*78
$GPGGA,174309.00,5600.56125,N,03822.26864,E,1,04,4.73,150.8,M,12.5,M,,*59
$GPRMC,174310.00,A,5600.56107,N,03822.26918,E,1.552,,280217,,,A*74
[... 39 lines cut ...]
$GPRMC,174322.00,A,5600.55887,N,03822.26453,E,1.466,,280217,,,A*73
$GPGGA,174322.00,5600.55887,N,03822.26453,E,1,04,4.71,151.9,M,12.5,M,,*58
$PFLAU,1,1,2,1,1,-72,2,287,1093*7C
$PFLAA,1,338,-1040,287,2,FCA567,,,,,1*19
$GPRMC,174323.00,A,5600.55896,N,03822.26494,E,1.254,,280217,,,A*7E
$GPGGA,174323.00,5600.55896,N,03822.26494,E,1,04,4.71)▒▒r▒bj▒12.5,M,,*5C
$GPRMC,174324.00,A,5600.55906,N,03822.26528,E,0.804,,280217,,,A*79
$GPGGA,174324.00,5600.55906,N,03822.26528,E,1,0$PFLAU,1,1,2,1,1,-72,2,287,1094*7B
$PFLAA,1,338,-1040,287,2,FCA567,,,,,1*19
4,4.71,151.8,M,12.5,M,,*5A
$GPRMC,174325.00,A,5600.55900,N,03822.26489,E,0.684,,280217,,,A*72
$GPGGA,174325.00,5600.55900,N,03822.26489,E,1,04,4.71,152.4,M,12.5,M,,*58
$PFLAU,1,1,2,1,1,-74,2,286,1116*77
$PFLAA,1,307,-1073,286,2,FCA567,,,,,1*14
$GPRMC,174326.00,A,5600.55889,N,03822.26455,E,0.948,,280217,,,A*7F
$GPGGA,174326.00,5600.55889,N,03822.26455,E,1,04,4.71,152.5,M,12.5,M,,*5B
$GPRMC,174327.00,A,5600.55849,N,03822.26306,E,0.948,,280217,,,A*73
$GPGGA,174327.00,5600.55849,N,03822.26306,E,1,04,4.70,154.1,M,12.5,M,,*54
$PFLAU,1,1,2,1,1,-74,2,284,1115*76
$PFLAA,1,307,-1072,284,2,FCA567,,,,,1*17
$GPRMC,174328.00,A,5600.55857,N,03822.26318,E,0.694,,280217,,,A*72
$GPGGA,174328.00,5600.55857,N,03822.26318,E,1,04,4.70,154.5,M,12.5,M,,*5F
$PFLAU,1,1,2,1,1,-76,2,284,1131*72
$PFLAA,1,273,-1098,284,2,FCA567,,,,,1*11
```

lyusupov, 2017-03-02:
Fixed in https://github.com/lyusupov/SoftRF/commit/c57a812f7b8df99402bc9689fd88598591123adc

closed 2017-03-02 by lyusupov