lyusupov/SoftRF #2: getting random different position from FLARM Mouse

biturbo, 2017-04-06:
Hi, I have build your system and I have a FLARM Mouse nearby (1 meter), but the position from the FLARM is jumping all around and its way off, do you have an idea whats this could be?

$PFLAU,1,1,2,1,1,45,2,6170,398993,DD50E2*3C
$PFLAA,1,282131,282131,6170,2,DD50E2,,,-228,,1*23
$PFLAU,1,1,2,1,1,45,2,6170,398993,DD50E2*3C
$PFLAA,1,282131,282131,6170,2,DD50E2,,,-228,,1*23
$PFLAU,1,1,2,1,1,175,2,6589,175154,DD50E2*0E
$PFLAA,1,-174487,15265,6589,2,DD50E2,,,558,,1*1D
$PFLAU,1,1,2,1,1,175,2,6589,175154,DD50E2*0E
$PFLAA,1,-174487,15265,6589,2,DD50E2,,,558,,1*1D
$PFLAU,1,1,2,1,1,-144,2,1658,242956,DD50E2*24
$PFLAA,1,-196556,-142806,1658,2,DD50E2,,,808,,1*0D
$PFLAU,1,1,2,1,1,-144,2,1658,242956,DD50E2*24
$PFLAA,1,-196556,-142806,1658,2,DD50E2,,,808,,1*0D
$PFLAU,1,1,2,1,1,-144,2,1658,242956,DD50E2*24
$PFLAA,1,-196556,-142806,1658,2,DD50E2,,,808,,1*0D

biturbo, 2017-04-07:
adding some findings, with the OGN client (rtlsdr-ogn) I am getting correct position data for the same FLARM Mouse source.

APRS <- FLRDD50E2>APRS,qAR:/063641h4735.00N\00852.71E^000/000/A=001391 !W68! id22DD50E2 +020fpm +0.0rot 48.0dB 0e -13.4kHz gps1x1
0.567sec:868.187MHz:   8:2:DD50E2 063647: [ +47.58344,  +8.87865]deg  424m  +0.1m/s   0.0m/s 000.0deg  +0.0deg/sec 1 01x01m 00f_-13.46kHz 48.2/60.0dB/0  0e    0.8km 060.3deg  -1.2deg   ?
0.573sec:868.187MHz:   8:2:DD50E2 063648: [ +47.58344,  +8.87865]deg  424m  -0.1m/s   0.0m/s 000.0deg  +0.0deg/sec 1 01x01m 00f_-13.41kHz 48.5/61.0dB/0  0e    0.8km 060.3deg  -1.2deg + ?
0.072sec:868.387MHz:   8:2:DD50E2 063649: [ +47.58344,  +8.87865]deg  424m  -0.1m/s   0.0m/s 000.0deg  +0.0deg/sec 1 01x01m 01f_-13.42kHz 48.2/62.0dB/0  0e    0.8km 060.3deg  -1.2deg + ?
0.698sec:868.186MHz:   8:2:DD50E2 063650: [ +47.58344,  +8.87865]deg  423m  -0.1m/s   0.0m/s 000.0deg  +0.0deg/sec 1 01x01m 00f_-13.59kHz 48.5/60.0dB/0  0e    0.8km 060.3deg  -1.2deg   ?
0.991sec:868.387MHz:   8:2:DD50E2 063650: [ +47.58344,  +8.87865]deg  423m  -0.1m/s   0.0m/s 000.0deg  +0.0deg/sec 1 01x01m 01f_-13.42kHz 48.0/61.5dB/0  0e    0.8km 060.3deg  -1.2deg   ?
0.582sec:868.187MHz:   8:2:DD50E2 063651: [ +47.58344,  +8.87865]deg  423m  -0.1m/s   0.0m/s 000.0deg  +0.0deg/sec 1 01x01m 00f_-13.43kHz 48.2/61.0dB/0  0e    0.8km 060.3deg  -1.2deg + ?
0.548sec:868.187MHz:   8:2:DD50E2 063652: [ +47.58344,  +8.87865]deg  423m  -0.1m/s   0.0m/s 000.0deg  +0.0deg/sec 1 01x01m 00f_-13.47kHz 48.2/60.0dB/0  0e    0.8km 060.3deg  -1.2deg   ?
0.063sec:868.387MHz:   8:2:DD50E2 063653: [ +47.58344,  +8.87865]deg  423m  -0.1m/s   0.0m/s 000.0deg  +0.0deg/sec 1 01x01m 01f_-13.41kHz 48.2/61.0dB/0  0e    0.8km 060.3deg  -1.2deg + ?
0.904sec:868.387MHz:   8:2:DD50E2 063654: [ +47.58344,  +8.87865]deg  423m  -0.1m/s   0.0m/s 000.0deg  +0.0deg/sec 1 01x01m 01f_-13.45kHz 48.2/61.0dB/0  0e    0.8km 060.3deg  -1.2deg + ?
0.788sec:868.186MHz:   8:2:DD50E2 063655: [ +47.58344,  +8.87865]deg  423m  +0.1m/s   0.0m/s 000.0deg  +0.0deg/sec 1 01x01m 00f_-13.62kHz 48.5/61.5dB/0  0e    0.8km 060.3deg  -1.2deg   ?
0.092sec:868.387MHz:   8:2:DD50E2 063655: [ +47.58344,  +8.87865]deg  423m  +0.1m/s   0.0m/s 000.0deg  +0.0deg/sec 1 01x01m 01f_-13.36kHz 48.2/63.0dB/0  0e    0.8km 060.3deg  -1.2deg   ?
0.652sec:868.187MHz:   8:2:DD50E2 063656: [ +47.58344,  +8.87865]deg  423m  +0.1m/s   0.0m/s 000.0deg  +0.0deg/sec 1 01x01m 00f_-13.46kHz 48.2/61.5dB/0  0e    0.8km 060.3deg  -1.2deg   ?

lyusupov, 2017-05-07:
FLARM units were updated with new firmware prior to April 1st. And so OGN guys did (0.2.6).

It is very likely that air data format/encryption were affected: https://github.com/creaktive/flare/issues/5

I don't have a FLARM unit locally. Could you submit some raw data captured with your OGN receiver?

lyusupov, 2017-05-11:
... or alternatively you may enable "private NMEA" setting in SoftRF, pick few $PSRFI strings and put them here in the thread.
But make sure first that your SoftRF UTC timestamp is accurate. Otherwise decoding of the raw data packet have no sense.

lyusupov, 2017-08-06:
Issuer of the ticket did not respond on request to provide additional details.

Local tests with FLARM unit and remote feedback from builder(s) of the SoftRF worldwide did NOT show any significant incompatibility with FLARM protocol at this time. 

So this issue is resolved with "unable to reproduce" status.

closed 2017-08-06 by lyusupov