lyusupov/SoftRF #3: Legacy emulator NullPointerException

ZaxCZ, 2017-10-01:
Hello, I am trying the prime version. When I try running Emulator.py, I am getting the following error. Any idea what might be causing it or how can I debug it?

```
pts/emulator/Emulator.py" && exit        <
legacy_emulator: bridge_host = 192.168.4.255 , bridge_port = 12390 , xcsoar_host = localhost , xcsoar_port = 4353
java.lang.NullPointerException: throw with null exception
```

lyusupov, 2017-10-01:
For me it looks like you have some general issue of QPython running on your particular Android device.
Try to run any other python script under QPython at first.
Then try to run the Emulator another version of QPython, or Android, or smartphone/tablet device. 

I have no such issue with Android 5.0 and QPython 1.2.5

Please, report, when your issue will become faded away.

ZaxCZ, 2017-10-04:
It is working now, probably some issue with my phone. It's weir that restart didn't fix that before.

closed 2017-10-04 by ZaxCZ