lyusupov/SoftRF #6: multiple protocols

geofrancis, 2017-11-22:
From what I can tell from the documentation this system is compatible with a lot of different protocols but can only do one at a time, would it be possible for it to cycle through them all every few seconds to get a more complete picture of whats around.

is it possible to use the lora and NRF905 on the same board and just switch between them rather than having to use 2 separate nudemcu dev boards?

geofrancis, 2018-01-26:
Would this be something you would be interested in doing?

lyusupov, 2018-03-05:
New feature requests do not subject for taking into consideration unless they are coming from co-developers.

> ... is it possible to use the lora and NRF905 on the same board ...

Not with current shield's PCB design.
But this makes no sense since "SoftRF LoRa" is able to manage all the protocols by itself.

However, if compatibility with "legacy" is the most critical factor - I still recommend for builders to stay with NRF905 module, because "LoRa" currently has 2 known minor performance issues with "legacy".

closed 2018-03-05 by lyusupov