lyusupov/SoftRF #7: Not working

MaUwJo, 2018-02-09:
Hi,

I have installed it to my board with the command:

esptool.py --port /dev/ttyUSB2 write_flash 0x00000  SoftRF-firmware-v1.0-rc1.bin 
Connecting...
Erasing flash...
Wrote 406528 bytes at 0x00000000 in 39.3 seconds (82.8 kbit/s)...

Leaving...
----
I can't connect to the web server on 192.168.4.1:

Serial output is:

--- Miniterm on /dev/ttyUSB2  38400,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
....
`Failed to mount file system
HTTP server has started at port: 80
␃�氰OYe}�2�1�S�}��;���`p�X��h2�S␝��Y�8�␝␝C;�
Reset reason: 3
Software Watchdog
Free heap size: 34504
Fatal exception:4 flag:3 (SOFT_WDT) epc1:0x400005cb epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000028 depc:0x00000000

Warning! EEPROM magic mismatch! Loading defaults...
EEPROM version: 90
SX1276 RFIC is detected.

Failed to mount file system
HTTP server has started at port: 80`

Any idea, what I'm doing wrong?

Thanks

lyusupov, 2018-02-09:
On rc1 the WebUI is available on 192.168.1.1

MaUwJo, 2018-02-09:
The Wifi vanishes each time I get this:
Fatal exception:4 flag:3 (SOFT_WDT) epc1:0x400005cb epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000028 depc:0x00000000
on the serial line.
IP-Address ist 192.168.4.2 with a gateway of 192.168.4.1

Thanks for the support.

lyusupov, 2018-02-09:
try out with GNSS modules been detached.

Use "code" markup tag to bracket log's data. It is unreadable otherwise.

MaUwJo, 2018-02-14:
Maybe my problem is that I don't have the GNSS module attached. :-) 
Is there an other firmware for the "SoftRF Prime Edition" with no GNSS attached?

Thanks

lyusupov, 2018-02-20:
Reproduction of the report on Prime with"LoRa" :

```
pi@raspberrypi:/run/tmp $ wget https://github.com/lyusupov/SoftRF/raw/master/software/firmware/binaries/SoftRF-firmware-v1.0-rc1.bin
--2018-02-20 09:52:12--  https://github.com/lyusupov/SoftRF/raw/master/software/firmware/binaries/SoftRF-firmware-v1.0-rc1.bin
Resolving github.com (github.com)... 192.30.253.112, 192.30.253.113
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/lyusupov/SoftRF/master/software/firmware/binaries/SoftRF-firmware-v1.0-rc1.bin [following]
--2018-02-20 09:52:13--  https://raw.githubusercontent.com/lyusupov/SoftRF/master/software/firmware/binaries/SoftRF-firmware-v1.0-rc1.bin
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.12.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.12.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 405840 (396K) [application/octet-stream]
Saving to: ‘SoftRF-firmware-v1.0-rc1.bin’

SoftRF-firmware-v1.0-rc1.bin      100%[===========================================================>] 396.33K   932KB/s    in 0.4s

2018-02-20 09:52:13 (932 KB/s) - ‘SoftRF-firmware-v1.0-rc1.bin’ saved [405840/405840]

pi@raspberrypi:/run/tmp $ ls -l SoftRF-firmware-v1.0-rc1.bin
-rw-r--r-- 1 pi pi 405840 Feb 20 09:52 SoftRF-firmware-v1.0-rc1.bin

pi@raspberrypi:/run/tmp $ ./esptool-2.2.1/esptool.py --port /dev/ttyUSB0 erase_flash
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 14.5s
Hard resetting...

pi@raspberrypi:/run/tmp $ ./esptool-2.2.1/esptool.py --port /dev/ttyUSB0 write_flash 0x00000 SoftRF-firmware-v1.0-rc1.bin
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 405840 bytes to 285743...
Wrote 405840 bytes (285743 compressed) at 0x00000000 in 26.5 seconds (effective 122.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

```

Boot log of the Prime in separate **minicom**'s window:
```
Reset reason: 6
External System
Free heap size: 32104
Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00

Warning! EEPROM magic mismatch! Loading defaults...
EEPROM version: 90
SX1276 RFIC is detected.

Failed to open cl_conf.txt.
No WiFi connection information available.
Wait for WiFi connection.
....................
Can not connect to WiFi station. Go into AP mode.
Setting soft-AP configuration ... Ready
Setting soft-AP ... Ready
IP address: 192.168.1.1
UDP server has started at port: 12389
HTTP server has started at port: 80

```
**Ok. Everything is just fine!**

Back to original report:

> 
> Hi,
> I have installed it to my board with the command:
> esptool.py --port /dev/ttyUSB2 write_flash 0x00000 SoftRF-firmware-v1.0-rc1.bin
> Connecting...
> Erasing flash...
> Wrote bytes at 0x00000000 in 39.3 seconds (82.8 kbit/s)...

The firmware's file size here is **406528**.
And it has to be (see my log above):  **405840** 

So, go download the firmware file correct way and repeat the flashing procedure.

**Resolution: Invalid**

closed 2018-02-20 by lyusupov

MaUwJo, 2018-02-22:
Thank you very much. I used an old version of esptool.py
markus@matrix:~$ cat /home/markus/hack/nodemcu/esptool/esptool.py| grep Cop
# Copyright (C) 2014 Fredrik Ahlberg
markus@matrix:~$ cat  /usr/local/bin/esptool.py | grep Cop
# Copyright (C) 2014-2016 Fredrik Ahlberg, Angus Gratton, Espressif Systems (Shanghai) PTE LTD, other contributors as noted.

I installed the new one with 

pip2.7 install  esptool

You wrote:
At first I've made a quick port of the FANET onto NodeMCU (ESP8266) and Semtech SX1276 radio
... can you upload it?

Thank you very much.