lyusupov/SoftRF #10: Software Compile Fails

kinnonk, 2018-02-14:
Hi 

Trying to compile the latest source - but getting the following errors - any ideas ?

Arduino: 1.8.5 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

Protocol_FANET.cpp:60: error: 'const rf_proto_desc_t {aka const RF_PROTOCOL}' has no non-static data member named 'tx_interval_min'

 };

 ^

exit status 1
'const rf_proto_desc_t {aka const RF_PROTOCOL}' has no non-static data member named 'tx_interval_min'

lyusupov, 2018-02-14:
Build options: Arduino Core for ESP8266 v2.4.0 , **lwIP v1.4 prebuilt**

lyusupov, 2018-02-14:
With Arduino 1.8.5 on **Debian** I am unable to confirm any build issues- 
current build works fine to me both for NodeMCU and ESP32.

Resolution: **Unable to reproduce**.

closed 2018-02-14 by lyusupov