lyusupov/SoftRF #14: Question about LED indicators

ljozsa, 2018-02-24:
Hi,

when I power up the device I see a self test of the LED's and it seems to be working well for the LED ring. However, LED's in vertical LED bar are blinking synchronously all the time and when the test finishes, all LED's are lit with red color. How are the LED's supposed to be interconnected? Right now I have connected all plus inputs on one wire, all ground inputs on another wire and all DIN inputs on the last wire. Is that correct?

Also, after self test, the LED's in the LED ring are illuminated a little bit by RGB colors. Is it correct behavior? 

What is the purpose of the LED bar? What should it indicate?

Many thanks,
L.

lyusupov, 2018-02-24:
vertical LED strip is extension of the ring.
each DO is connected to DI of the next one in sequence.
 
They are status LEDs. Purpose of each is indicated on the faceplate: https://github.com/lyusupov/SoftRF/blob/master/case/v4/SoftRF-Case-v4-Face-Plate.stl
Order is the same or close to that the FLARM has.

closed 2018-02-24 by lyusupov

lyusupov, 2018-02-24:
> LED's in the LED ring are illuminated a little bit by RGB colors

Yes, this is intentional backlit for dark daytime or shade when it is hard to see the arrow on faceplate.