lyusupov/SoftRF #15: No FLARM sentences

kinnonk, 2018-03-15:
Hi, 

This may be an issue with the settings I am using but when connected to the Soft-RF module I get what seems to be normal NMEA sentences, but no FLARM sentences.    Consequently when I check the status of the device I see GPS fix rather than GPS fix; FLARM in XCSoar.     Am I doing something wrong - using XCSoar with GNSS and Legacy protocols switched on with Legacy/Normal mode ?

Thanks in advance for help !

Regards, 

Kenny

lyusupov, 2018-03-16:
When submitting an issue, please, supply screenshots of SoftRF's:
-  status page ;
- settings page;

$PFLAU and $PFLAA FLARM NMEA sentencies are being emitted over UART/BT only when:
- SoftRF has valid GNSS fix, and
- "NMEA legacy" is enabled in settings, and
- SoftRF is receiving valid RF packets of selected radio protocol from a neighbour proximity awareness unit.

kinnonk, 2018-03-16:
Thanks Linar !

Got it sorted now - great little project by the way !

Thanks 

Kenny

closed 2018-03-16 by lyusupov

ZaxCZ, 2018-07-11:
What are the correct/recommended XCSoar settings (i.e. what driver should I use)? I keep getting "no data" in device manager, but I can see it's receiving - looks like [this](https://i.imgur.com/5fNKcUN.png) (might be a wrong baud rate, I'm not sure what format to expect, this one looked closest to something readable).

As you said, this might be caused by not having another awareness device nearby. In the next few days, I will hopefully test it with FLARM and also OGN receiver.

_I thought this was kinda related and not worth opening another ticket for, so I ask this here. Awesome work, Linar! _

lyusupov, 2018-07-11:
your report looks like a  duplicate of this https://github.com/lyusupov/SoftRF/issues/11

ZaxCZ, 2018-07-12:
Thank you, you are correct. I have set the baud rate to 38400, one stop bit, no parity bits, now I'm getting FLARM and GPS in XCSoar, however, I'm getting multiple `bad parity of decoded packet: 1` and some NMEA sentences, including my location (I have left out the GNGLL sentence in the screenshot for obvious reasons). Looks like [this](https://i.imgur.com/uqK1nxx.png).

lyusupov, 2018-07-12:
"bad parity of decoded packet" typically happens with "Legacy" protocol when SoftRF is out of time sync with GNSS date & time.

For information on GNSS module set-up read:
1. [this](https://github.com/lyusupov/SoftRF/issues/13#issuecomment-368247525)
2. and [this](https://github.com/lyusupov/SoftRF/issues/1#issuecomment-281290554)

What particular MCU do you use? and what particular GNSS module?

lyusupov, 2018-07-13:
Ok,

1. You'd better open  a separate GitHub ticket rather than re-using this one  
2.  With your report I am uncertain do you test it against a FLARM or another SOFTRF unit?
If this is the second case - you have to be 100% sure that the partner instance has valid and accurate GNSS time & date. Check it"s status WebUI page. If any of partner's is more than 64 seconds out of sync - "legacy" encryption/decryption will not work properly.
3. SoftRF ignores most of NMEA sentencies except GxRMC and GxGGA. **These two are mandatory**. Use U-center and get rid of all other sentencies except these two.
4. There were no reports to me about non-compliance with V6 after this update: https://github.com/lyusupov/SoftRF/commit/2491a808dfc8ac92e9dc24fbd9012559ba30d926
5. There are plenty of active SoftRF builders/users in Czech Republic. At least two of them are GitHub users ( https://github.com/ljozsa https://github.com/jharting ) . Why don't you ask them if they have same issues you are sufferng from, directly and in your native language?

lyusupov, 2018-07-13:
Ok. Thanks! Now it makes more sense...

So,  you have NRF module plugged in, no any FLARM sources around, GNSS module got a fix, and just after that it stats to emit "bad parity" messages, right?

Ok. Check WebUI status page. See if the Radio is detected as NRF905. If not - check every soldering work you did and replace NodeMCU with another one. Same thing with the NRF module.

If WebUI says - the module is in, go to to the Settings, enable Private messages, save and restart.
Capture output of $PSRFI messages (raw packet's hexadecimal data) and copy it here in the thread, or send to me by e-mail.