lyusupov/SoftRF #17: Same address name in all devices

CT2IUV, 2018-04-15:
Hi Linar,

I program 3x esp32 to replace all 8266 but now all the devices get the same ID, in my case softrf-a4ae30, so for the open glider network all 3 devices is the same device because all have the same ID.

I would  like to change the ID, to the Glider ICAO registration Number... I already thy but without success,

Best regards

lyusupov, 2018-04-22:
Try out [this commit](https://github.com/lyusupov/SoftRF/commit/044ca314ef73a5b9ebc505118ec6298cecef734a)

Report if it works for you.

CT2IUV, 2018-04-22:
Hi Linar,

Yes it works.

I have a problem with the use off the bluetooth but is my issue, is
someting missing in my library i think. i will check.

But the important for you now is, all my 3 devices now have diferent ID.

I edit all the important code for garanty all devices have diferent ID
using the mac address and it works.

Thanks

Regards

Nuno Freitas

closed 2018-04-23 by lyusupov