lyusupov/SoftRF #18: Size issue on ESP32 with RC3 - 102% on Heltec ESP32

GithubReinart, 2018-05-14:
Hello Linar,

thanks for your great work. 

Trying out SoftRF on ESP8266 and ESP32 HW since April.
WIth RC3 I am observing size issues now when building in Arduino environment following Instruction concerning board setup and the specific libraries.
While RC2 consumed only about 86% of the maximum space for Heltec ESP32 board RC3 now uses 1337326 Bytes (102%) ... Maximum is 1310720 Bytes.

Do you have hint for basic Arduino setup or other libraries?

KInd Regards
Manfred

lyusupov, 2018-05-14:
Manfred,

Keep monitoring [official build instructions](https://github.com/lyusupov/SoftRF/tree/master/software/firmware/source) - they are typically become updated with each significant change of the procedure.
You need to switch onto "Minimal SPIFFS" partition scheme.

A warning at build time may still appear. It can be safely ignored.
For details, read [this](https://github.com/espressif/arduino-esp32/pull/1302) and [that](https://github.com/espressif/arduino-esp32/issues/1378)
In order to get rid of the annoying warning, fix your local Arduino ESP32 Core's **boards.txt** using 
my [Travis CI](https://github.com/lyusupov/SoftRF/blob/master/.travis.yml) script as an example.

GithubReinart, 2018-05-14:
Hello Linar,

Thanks for the quick reply - I already set „Minimal SPIFFS“ partition scheme before sending the message.
Will also check the other information provided.

Many Thanks
Manfred

lyusupov, 2018-05-14:
One more comment regarding Heltec:
you will not get a picture on the Heltec's OLED until your ESP32 Core's I2C will not be updated onto "experimental" version been developed by Chuck Todd.
 Follow [these instuctions]( https://desire.giesecke.tk/index.php/2018/04/20/how-to-use-stickbreakers-i2c-improved-code/ ).

closed 2018-05-14 by lyusupov