lyusupov/SoftRF #21: Compile failed

LilyGO, 2018-07-07:
I installed all the libraries, but the compilation failed, of course, I also installed the configuration file.
*********************************************
Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\15840\AppData\Local\Temp\arduino_cache_468712\core\core_espressif_esp32_esp32_PSRAM_disabled,PartitionScheme_default,FlashMode_dio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,DebugLevel_none_4cfcc43f8cdc37390bbcfde39fb1d404.a
libraries\Wire\Wire.cpp.o:(.bss.Wire1+0x0): multiple definition of `Wire1'

sketch\Platform_ESP32.cpp.o:(.bss.Wire1+0x0): first defined here

C:\Users\15840\AppData\Local\Temp\arduino_build_811450/arduino.ar(HardwareSerial.cpp.o):(.bss.Serial1+0x0): multiple definition of `Serial1'

sketch\Platform_ESP32.cpp.o:(.bss.Serial1+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board ESP32 Dev Module.
***********************************************
Then I burned the bin file on T-Beam, but the serial port information is garbled.
[image]

lyusupov, 2018-07-07:
You are using wrong destination addresses.<br>
SoftRF.ino.bin should go into 0x10000<br>
SoftRF.ino.partitions.bin should go into 0x8000<br>

For reference, read **firmware.txt** file that comes inside the firmware's zipfile.

lyusupov, 2018-07-07:
And I am curious, why are you using 26 MHz crystal setting?

closed 2018-07-09 by LilyGO