lyusupov/SoftRF #23: Qpython Can not run

LilyGO, 2018-07-09:
## First Qpython
### I installed Qpython on Google _play, but the interface is inconsistent. Maybe it's a new version.
[image]
## Second no main.py
### Then I transferred the file to my phone and ran it, but there was no main.py.
[image]
## Finally ,lib
### This python library, I downloaded it. But I don't know if the installation was successful.

LilyGO, 2018-07-09:
[image]

lyusupov, 2018-07-09:
This is not a QPython support site.
Known good version of  QPython that works with the Emulator is 1.2.5

**Emulator** is a deprecated software. It only has sense for GNSS-less **Prime Edition** in Bridge mode.
**Prime Edition** is a part of Archive now.

All current GNSS or UAV-capable SoftRF Editions do operate in Normal mode and thus they do not need Emulator anymore.

LilyGO, 2018-07-10:
What you mean is that the SoftRF-firmware-v1.0-rc4-ESP32 firmware does not require an Emulator. Can continue to work normally, right?

lyusupov, 2018-07-10:
Most people who deal with SoftRF project are pilots.
They are typically already familiar with either FLARM, OGN Tracker or PilotAware.
So they mutually understand basic principles of the technology.

From very beginning (v 0.9) SoftRF firmware has WebUI mode selector in 'Settings' area.
If target device (particular SoftRF Edition) has built-in GNSS source - this mode selector is to be set in "Normal" position.
If target device does NOT have GNSS - the only option applicable is "Bridge" mode, when all data processing is off-loaded from the SoftRF device onto an external Androd or Raspberry Pi host.

lyusupov, 2018-07-11:
[Quick start](https://github.com/lyusupov/SoftRF/wiki/Prime-Edition-MkII.-Quick-start) section is added into the Wiki. Hope that this will help you and other beginners.

closed 2018-07-12 by lyusupov