lyusupov/SoftRF #29: Self compiled FW for TTGO T-Beam - no GPS

Miirek, 2018-09-06:
I'm not sure whether it is an issue but I've compiled project and uploaded it to my T-Beam - but no GPS detected, while when using your precompiled rc4 everything works. 

I'm using Arduino IDE 1.6.13 @ OSX 10.13, environment set up as you described here. Compilation has no issues.

JuergenLeber, 2018-09-06:
Looks like the device recognition is not working properly. As a quick workaround just change the default model in SoftRF.ino:

...
hardware_info_t hw_info = {
  .model    = SOFTRF_MODEL_PRIME_MK2,
...

JuergenLeber, 2018-09-06:
Didn't try BT yet, will do the next days. I also noticed that I couldn't flash anymore when BMP280 was connected. After disconnecting it it worked again. Just in case you run into that problem :-)

Miirek, 2018-09-06:
I've read about it. In my case, with BME280 soldered, reflash (via arduino IDE) works  without problem.

lyusupov, 2018-09-06:
Guys, while doing the build from sources you have to consider that:
1) development tree can be at "bleeding edge" state. No any guarantee that it works at any particular moment ;
2) source code build instructions can be not up to date.

Right now, it is the 2nd case - to get more or less working code for ESP32 you have to build it against very first stable build of Arduino Core: https://github.com/espressif/arduino-esp32/releases

I will update ESP32 build instructions on time permitting basis.

I don't see any issues in this reports expect that the reporter had built the code against wrong Arduino Core revision.

lyusupov, 2018-09-06:
**Resolution**: unable to reproduce

closed 2018-09-06 by lyusupov