lyusupov/SoftRF #30: AirConnect protocol

JuergenLeber, 2018-09-09:
Just tried AirConnect with "Air Navigation Pro" app but didn't succeed. I defined AirConnect as active and server gets started. Port 2000 is working and delivering NMEA data. But I think the protocol is not fully correct. May I help here or are you already working on this? Not that we do double work.

JuergenLeber, 2018-09-11:
I'll have a look into it, please assign to me!

JuergenLeber, 2018-09-20:
Was a problem with a missing "\n". The AirConnect protocol waits for four digits and then starts sending the NMEA output if password is correct. Implemented in PR #34 .

lyusupov, 2018-09-21:
Try out this commit: https://github.com/lyusupov/SoftRF/commit/c0038e10c2e84c34e787c2067237aaaabba2c7e8

It works for me:

[image]

<br>

[image]

And, please, double check that no any **CR**s or **LF**s after **PASS?** and **AOK** are actually in use in genuine AirConnect. Because I would prefer to see a better look in the monitor window.

JuergenLeber, 2018-09-21:
In genuine AirConnect no CRs or LFs are used. It only sends "PASS?" and waits for either LF or exactly four bytes of data without LF and then sends "AOK" if password is/was correct. After that directly the NMEA data is sent. So I don't know if your version will work 100% as it doesn't count the four bytes but instead has a timeout. I'll try!

lyusupov, 2018-09-21:
My opinion that this is an Air Navigation Pro's issue.
SkyDemon is smart enough - when it already sees NMEA sentences in the input stream - there is no need to enter pass codes anymore.
XCSoar does not care about handshake at all.

**I am definitely not going to break compatibility** with XCSoar, LK8000 and SeeYou as most of glider pilots are using them on daily basis, **just in order to make AirNav Pro users happy**.

JuergenLeber, 2018-09-21:
I'll try with Air Nav Pro and most probably it's supposed to work with the timeout solution even if it's not the correct AirConnect protocol standard. So I can't tell you where the issue is exactly located. Let's find a solution to make all app's users happy.

JuergenLeber, 2018-09-21:
Works with Air Nav Pro. But ANP still says "Bad GPS for navigation", probably something with GPRMC vs. GNRMC. But that's another problem.

closed 2018-09-21 by JuergenLeber