lyusupov/SoftRF #31: GDL90 Ownship Geometric Altitude

maehtricks, 2018-09-10:
First of all: thanks for your fantastic work.

I've noticed an issue with RC5 regarding GDL90 Ownship Geometric Altitude. With RC4 the altitude was shown correctly in SkyDemon, but with RC5 it's off.

GDL90 spec states that it has to be reported as relative to the WGS84 ellipsoid, with RC5 it looks like MSL alt is send.

- WebUI: 91m - correct alt
- XCSoar (FLARM): 138m - too high, looks like ellipsoidal height is send instead of MSL alt
- SkyDemon (GDL90): 130ft - too low, looks like MSL alt is send instead of ellipsoidal height

I'm using Prime MK2 Hardware.

lyusupov, 2018-09-11:
Please, provide, being taken at earth surface level, screenshots (width <=800 pixels) of:
1) status page ;
2) settings page ;
3) 'monitor' window of XCSoar's device menu (in portrait mode) with at least one full second burst of GNSS NMEA sentences.

maehtricks, 2018-09-12:
Alright, I did some testing this morning. My hardware is t22_v07.

I took screenshots from RC4 and RC5:

1. Status
2. Settings
3. SkyDemon (GDL90)
4. Avare external I/O plugin (GDL90)
5. XCSoar (NMEA)

### RC4
status
[image]
settings
[image]
skydemon
[image]
avare
[image]
xcsoar
[image]

### RC5
status
[image]
settings
[image]
skydemon
[image]
avare
[image]
xcsoar
[image]

lyusupov, 2018-09-12:
Ok. Thank you! 

**You report is a mix of two issues**:
1) in both RC4 and RC5 there was a minor bug in GDL90 geometric altitude reporting. Fixed in https://github.com/lyusupov/SoftRF/commit/cf5d37eebaf47fbcfb1fa40dc7fb37fa9d81b3d0
2) you have so-called ["heavy counterfeit" GNSS module](http://ttgobbs.com/viewthread.php?tid=17131&rpid=155152&ordertype=0&page=2#pid155152) soldered into your T-Beam.
 
**Indicators of that**:

[image]

and

[image]

Issue with these GNSS modules is that they always report "0.0" as geoidal separation value.
It was recently first discovered and reported by [Pawel Jalocha](https://github.com/pjalocha) .
 
Most of modern GNSS ICs are reporting AMSL altitude in their NMEA $GxGGA sentence: http://aprs.gids.nl/nmea/#gga
Followed by "geoidal separation value" - hight difference between WGS84 ellipsoid and EGM96 geoid.

We don't know for sure what is the "altitude" value reported by this "heavy counterfeit" GNSS chip, but it looks like an altitude relative to WGS84 ellipsoid.
Old GPSes (prior to year 2000 or so) were also reporting WGS84 altitude, **but they were suppling EMPTY geo separation field**.

This is an example for your latitude and longitude taken from [online calculator](https://www.unavco.org/software/geodetic-utilities/geoid-height-calculator/geoid-height-calculator.html):
```

Your Input Coordinates and GPS Height: 

Latitude = 49.63° N = 49° 37' 48" N 
Longitude = 8.61° E = 8° 36' 36" E 
GPS ellipsoidal height = 142.0 (meters) 
Geoid height = 48.035 (meters) 
Orthometric height (height above EGM96 geoid which approximates mean sea level) = 93.965 (meters) 

(note: orthometric height = GPS ellipsoidal height - geoid height) 
```

**RC5 has a workaround implemented against these "heavy counterfeit" ICs**. The algorithm of EGM96 based altitude adjustment was borrowed from XCSoar. So reported SoftRF RC5 WebUI AMSL altitude should be the same or very close to the value taken from XCSoar's navbox.
Due flash memory space constraints, geo separation value is avaraged over 2 degrees lat/lon "square" earth surface. Not perfect, but still acceptable for most aviation applications.

lyusupov, 2018-09-12:
I am curious:
1) why are you using 868.2 MHz for P3I ?

[image]

have you read this ? : https://github.com/lyusupov/SoftRF/wiki/Settings

2) Which way did get your RC5? Custom built?

[image]

It should report more that 4000000 bytes on officially released RC5 running on T-Beam board.

maehtricks, 2018-09-12:
Cool, thank you for the fix and the explanations. Is there a special setting to activate the altitude workaround in XCSoar? Right now it's showing the same alt (GPS ellipsoidal) in the alt gps box as in NMEA sentences with no correction.

1. I'm using the legacy protocol normally... so this was some weird setting from last night's testing.

2. I was experimenting as well with the GDL90 alt with my own build... but the screenshots were taken from the same source as your's. But you're right, with your build it shows more than 4000000 bytes. I'll have to check what's wrong with my setup.

maehtricks, 2018-09-12:
Checked this... XCSoar's correction just kicks in if it's not a valid double value. Since 0.0 is valid, no correction will be performed. I saw this in the code and tested it by playing back a modified nmea caputure. If I remove 0.0 the proper altitude is shown.

maehtricks, 2018-09-12:
Checked GDL90 Ownship Geometric Altitude report successfully with SkyDemon + Avare.

Thank you very much.

lyusupov, 2018-09-13:
> XCSoar's correction just kicks in if it's not a valid double value. Since 0.0 is valid, no correction will be performed. I saw this in the code and tested it by playing back a modified nmea caputure. If I remove 0.0 the proper altitude is shown.

maehtricks, 2018-09-13:
> 
> 
> @maehtricks
> 
> > XCSoar's correction just kicks in if it's not a valid double value. Since 0.0 is valid, no correction will be performed. I saw this in the code and tested it by playing back a modified nmea caputure. If I remove 0.0 the proper altitude is shown.
> 
> Try out this commit: [0ef8557](https://github.com/lyusupov/SoftRF/commit/0ef855703d8b2dc505cd97a5cafa68a85c581ebf)

Works perfect. Thank you again :)

[image]

lyusupov, 2018-09-13:
@maehtricks

Ok. Good !

Could you, please, take one more screenshot of NMEA sentences in XCSoar's "monitor" window  (<=800px) ?
 I need to take a look at it.

maehtricks, 2018-09-15:
Sorry for the delay... had some hardware issues. But here we go with BMP280 attached :) 
[image]
[image]

lyusupov, 2018-09-17:
@maehtricks:

Thank you!

I had to make sure that time/lat/lon of re-generated GGA sentence have same or close values & precision (digits after decimal point) as the parameters of pass-through RMC sentence.

[image]

lyusupov, 2018-11-19:
Although GDL90 spec states that altitude has to be reported as relative to the WGS84 ellipsoid,
most of EFBs and portable ADS-B devices are reporting MSL altitude.
See this info for details: https://github.com/cyoung/stratux/pull/749
SkyDemon seems to be the only exception (that follows the specs).

I am going to revert my previous change soon.
I will sacrifice SkyDemon's altitude accuracy in favour of all other EFBs accuracy.

maehtricks, 2018-11-20:
Hm, what do you think about adding it as option to be compatible to almost all EFBs on the market?

lyusupov, 2018-11-20:
No sense to add the GDL90 menu option for the sake of SkyDemon only.
Users of SkyDemon can use NMEA TCP ("AirConnect") instead.

Alternatively, you can push SD team to use the MSL, like all other EFBs do.

closed 2018-11-21 by lyusupov

ragflyer, 2021-08-18:
**Summary for other Skydemon users:**
In SoftRF settings set 'NMEA output' to 'TCP', then in SD settings menu -> Third party devices -> FLARM devices -> FLARM with Air COnnect to 'on', leave the key as is. Altitude should now be shown correctly.