lyusupov/SoftRF #35: No valid GNSS fix

chestnutbraun, 2018-09-22:
## Hardware

- [X]   Prime Mark II
- [ ]   Standalone
- [ ]   UAV

< photo of your hardware > (optional)
[image]

TTGO T-Beam Version: T22_V07 20180711

## Firmware version
1.0-rc4+ ESP32
<  screenshot of WebUI status page > (**required**)
[image]

## Firmware settings
[image]

<  screenshot of WebUI settings page > (**required**)

## Describe the bug (or ask a question)
In SoftRF I can not get a valid GNSS fix although the red LED on the board indicates GPS reception.
I then tried to update the firmware to 1.0-rc5 but the update failed. 1.0-rc4+ shows to be still the valid software version on the TTGO T-Beam
[image]
How can I get a valid GNSS fix in SoftRF? 

## To Reproduce

## Expected behavior

lyusupov, 2018-09-22:
I see this information in your status screenshot:

[image]

For me it means that the firmware does not see any NMEA sentences that should normally come out of T-Beam's GNSS module.

I don't know you skills level, so I assume that it is zero. It this case I would recommend you to work with your seller of the T-Beam board. Reminder, that my firmware is not responsible for any hardware issues with particular TTGO board, see [this disclaimer](https://github.com/lyusupov/SoftRF/wiki/Prime-Edition-MkII#disclaimer) for more details.

chestnutbraun, 2018-09-22:
Dear Linar, thank you for your time looking into this. Skills level is zero, I can measure voltage with a voltmeter and can solder. I was sure that it was not your firmware and I suspected that there is no original NEO-6M installed, I have read your disclaimer before ordering the board and hoped for the best...

lyusupov, 2018-09-23:
If you need some arguments prior to asking your seller for re-send or refund, try out either of those:

1) connect T-Beam with SoftRF to a PC and TTY terminal emulation software (PuTTY, HyperTerminal for Windows or minicom for Linux). Settings are 38400, 8N1

Regular boot log should look like this:

```
Reset reason: 0
No reset reason.
Free heap size: 4349856
No reset information available.
EEPROM version: 91
SX1276 RFIC is detected.
WARNING! Barometric pressure sensor is NOT detected.
INFO: TTGO T-Beam GPS module is detected.
WiFi config changed.
new SSID:
Hostname: SoftRF-c5d804
Wait for WiFi connection.
....................
Can not connect to WiFi station. Go into AP mode.
Setting soft-AP configuration ... Ready
Setting soft-AP ... Ready
IP address: 192.168.1.1
UDP server has started at port: 12389
HTTP server has started at port: 80
$GPRMC,042829.99,V,,,,,,,220818,,,N*79
$GPGGA,042829.99,,,,,0,00,99.99,,,,,,*63
$GPRMC,042831.00,V,,,,,,,220818,,,N*70
$GPGGA,042831.00,,,,,0,00,99.99,,,,,,*6A
$GPRMC,042832.00,V,,,,,,,220818,,,N*73
$GPGGA,042832.00,,,,,0,00,99.99,,,,,,*69
$GPRMC,042843.00,V,,,,,,,220818,,,N*75
$GPGGA,042843.00,,,,,0,00,99.99,,,,,,*6F

```

2)  Go onto TTGO's GitHub pages and use their GPS.ino sketch to test your T-Beam onto basic GNSS module operation:
 https://github.com/LilyGO/TTGO-T-Beam/tree/master/GPS

3) The onboard board GNSS module delivers serial data to GPIO12 pin of the ESP32. Unfortunately, GPIO12 is not available on PCB's pads for pin header. 
If you want, you may try to test if proper signals are available on Tx pad (No. 20) of the "NEO" module.
And you may trace if the signal is delivered to GPIO12 pin of the ESP32. 
Use [this schematics](http://ttgobbs.com/viewthread.php?tid=17131) (not up to date but still Ok) as a reference.

lyusupov, 2018-09-23:
Regarding to failure of updating the firmware over web:

As [it was written there](https://github.com/lyusupov/SoftRF/wiki/Firmware-update-(Web-method)#esp32), if you have not succeed with web firmware update method - proceed to [first-time firmware installation procedure](https://github.com/lyusupov/SoftRF/tree/master/software/firmware/binaries#esp32).

chestnutbraun, 2018-09-23:
The boot is as attached and it ends with the line: "HTTP server has started at port: 80"

Reset reason: 0
No reset reason.
Free heap size: 4350240
No reset information available.

EEPROM version: 91
SX1276 RFIC is detected.
Hostname: SoftRF-88bd74
Wait for WiFi connection.
....................
Can not connect to WiFi station. Go into AP mode.
Setting soft-AP configuration ... Ready
Setting soft-AP ... Ready
IP address: 192.168.1.1
UDP server has started at port: 12389
HTTP server has started at port: 80

Thank you again for all your help.

lyusupov, 2018-09-23:
Ok. This is an extra confirmation of [the statement](https://github.com/lyusupov/SoftRF/issues/35#issuecomment-423763102).

The NEO module is either:
1) does not emit NMEA sentences ; 
2) emits them, but on baud rate other than 9600 ;
3) emits NMEA, but they are not delivered to ESP32's GPIO12.

If I would have issue like your do, I would connect analog or DS oscilloscope to pin 20 (Tx) of "NEO" module.
If you are unable - work with LilyGO to send you a replacement board.

chestnutbraun, 2018-09-23:
Dear Linar,
now I was able to update to the latest firmware version 1.0-rc5 and now it is working! 
Again thank you very much for your support!

lyusupov, 2018-09-23:
If the ticket is not pertinent any more - please, close it.