lyusupov/SoftRF #39: FANET 1.1

3s1d, 2018-11-15:
Hi,

sorry for replying to your pull request so late (havn't seen it).

I'd like to draw your attention to the latest FANET 1.1 extension which will be publicly available early 2019. The main difference is that while not being airborne the device will emit type 7 (Ground Tracking) frames rather then type 1 packets. During flight every things remains unchanged.

Cheers,
Juergen

lyusupov, 2018-11-15:
Juergen,

thank you for the announcement!

Main purpose of my FANET protocol implementation is to avoid mid-air collisions of GA traffic with para- and hang- gliders. That's why it is limited to support of "type 1" data packets only.
I will probably stop transmitting "type 1" packets at all when the SoftRF unit is not moving along with it's aircraft.

I can see [here](https://github.com/3s1d/fanet-stm32/search?q=onGround&unscoped_q=onGround) that airborne state definition is out of the scopes of your open source firmware repo.
That is your take-off trigger logic?

    Regards,
    Linar.

3s1d, 2018-11-16:
For mid-air collisions you don't need to change anything.

The code is optimized for a module which is connected to flight instrument via UART. It's the flight instruments task to set the airborne state. 

Cheers,
Juergen

closed 2018-11-16 by 3s1d