lyusupov/SoftRF #44: RESET anomaly with ESP32 + RFM95W

bnsearle, 2018-12-17:
**ATTENTION** ! 

- Use this form to report bugs in released SoftRF firmware, hardware and web pages only ;
- Prior to asking a question, look into ["Closed issues"](https://github.com/lyusupov/SoftRF/issues?q=is%3Aissue+is%3Aclosed) area to see if it was answered before ;
- No any "ideas" or new feature requests, please ! New feature requests do not subject for taking into consideration unless they come from co-developers ;
- "Enhancements" of existing features are allowed only when you are ready to implement them by yourself ;

Recommended:
- screenshot or photo width - from 600 to 800 pixels ;
- aspect ratio - 4:3, 3:4, 16:9 or 9:16.
- substitute 'space' by 'x' character to mark a hardware checkbox

--------------------

## Hardware

- [ ]   Prime Mark II
- [ x]   Standalone
- [ ]   UAV

< photo of your hardware > (optional)

## Firmware version

<  screenshot of WebUI status page > (**required**)

/home/bryan/Pictures/Screenshot from 2018-12-17 17-25-38.png

## Firmware settings

<  screenshot of WebUI settings page > (**required**)
/home/bryan/Pictures/Screenshot from 2018-12-17 17-26-34.png

## Describe the bug (or ask a question)
Hardware: 
ESP32 Development Board, DOIT layout, wired to Hope RFM95W. Software latest SoftRF release 1.0-rc5, date 10/12/18, compiled on Arduino 1.8.5, settings as per instructions. Normal and Legacy mode settings.

Problem: 
If the RFM95W RESET is connected to the ESP32 GPIO-14 (as in Platform _ESP32.h) then after a power-up or EN button press then Radio = NONE and there are no RF traffic packets. If you then do a ‘soft’ restart i.e ‘settings/save and restart’ then we get Radio = SX1276 and packets are Tx and Rx, operation OK. However if I disconnect the RESET link then a power up or EN press work as expected.

Comments:
I would like to have the RESET connected on the PCB that I am designing that will use the WROVER module, because then there is the opportunity of introducing a protocol change ‘on the fly’ and/or the use of the watchdog feature  of the ESP32 which would enhance reliability (I think I may have seen hang-ups while soak testing, but I cannot be certain!).

## To Reproduce
see above

## Expected behavior
see above

bnsearle, 2018-12-21:
I have tried the latest version dated 20th, but it did not fix the issue for me. However I have made a fairly crude fix by inserting the following snippet in SoftRF.ino after line 164:

// crude fix for 'sx1276 not detected' issue
  if (resetInfo->reason == 0) { 
     RF_Shutdown();
     delay(1000);
     ESP.restart();
  }

  /* expedite restart on WDT reset */
.
.
...... etc

This reports a detected sx1276 (on the serial monitor and web app) after the unsuccessful detection on first power-up or EN button press. 

Now I have 2 units that always power-up correctly and communicate as expected - now at the full 20dBm level I see!   I have read that technically the EU limit is less than this (15?), but after antenna losses etc....

As an aside - it would be easier to debug if the NMEA stream came from the unused UART output i.e.  the UART that receives from the GPS module on GPIO22 ? I will attempt that in due course, so I can make that connection on the pcb,....

lyusupov, 2018-12-30:
The issue reported is **NOT CONFIRMED** on Standalone neither with:
* "stock" RC5 firmware binaries, nor
* firmware been built from most recent git source code.

Recommendations:
1) inspect your power and reset circits wiring ;
2) inspect your SPI bus wiring ;

**Resolution**: unable to reproduce.

closed 2018-12-30 by lyusupov