lyusupov/SoftRF #46: SoftRF WiFi connection problem on Kobo Touch 2

dadyeah, 2019-01-26:
Hi, this is probably a Kobo issue but I've had no luck finding a solution. SoftRF connects fine to an Android phone but will not connect to my Kobo Touch 2 (which connects ok to other WiFi sources)  . Any help much appreciated

lyusupov, 2019-01-26:
Issue was filed not in compliance with [this template](https://github.com/lyusupov/SoftRF/blob/master/.github/ISSUE_TEMPLATE/softrf-issue-template.md) (offered automatically for each new issue).

**Resolution**: invalid.

<br>
<br>

[image]

closed 2019-01-26 by lyusupov

dadyeah, 2019-01-27:
Fixed now with the help of DK in Texas on Gitter. Many thanks