lyusupov/SoftRF #54: BLE disconnects after some time and is not discoverable after that

smfoo, 2019-03-28:
## Hardware

- [x]   Prime Mark II
- [ ]   Standalone
- [ ]   UAV
- [ ]   UAT module
- [ ]   SkyView

[image]

## Firmware version

[image]

## Firmware settings

[image]

## Describe the bug (or ask a question)

After some time the BLE connection disconnects and also is not longer discoverable with a BT scanner like LightBlue or my flightapp. The device itself continues to work (other FLARM devices still are getting FLARM messages) and is reachable via wifi. If you restart it via the settings page it is discoverable again and works for some time. I've got two TTGO boards and they both have the described problem.

## To Reproduce

Use SoftRF in BLE mode. At some time it will stop working.

## Expected behavior

It should keep working and be discoverable as long as it is powered on.

lyusupov, 2019-03-31:
Thanks for report. I've updated this [user documentation area](https://github.com/lyusupov/SoftRF/wiki/Settings#built-in-bluetooth-5) to reflect the issue you have experienced with.

**I don't have iOS hardware & software. And have no plans to support it by myself.**
Thus, I can't even confirm if this is a bug or it is not.

Your issue is looking for me as indication that this ESP32 IDF & iOS compatibility bug is still remain active: https://github.com/nkolban/esp32-snippets/issues/545

Known good SoftRF and iOS connection method is Wi-Fi.
Known good  SoftRF and iOS connection protocols are GDL90 and NMEA TCP ("AirConnect").

I am closing this ticket with

**Resolution**: will not fix.

If you or someone else are ready to dig into this ESP32 and iOS compatibility issue - re-open this ticket and assign it onto yourself.

lyusupov, 2019-10-31:
Try [RC7](https://github.com/lyusupov/SoftRF/releases/tag/1.0-rc7) out.