lyusupov/SoftRF #55: No FLARM signal sent

GliderPil0t, 2019-04-03:
## Hardware

- [x]   Prime Mark II
- [ ]   Standalone
- [ ]   UAV
- [ ]   UAT module
- [ ]   SkyView

https://imgur.com/a/KgBDx7L

## Firmware version & settings

screenshot of WebUI settings & status page
https://imgur.com/a/J9BjCTn

## Describe the bug (or ask a question)
Hi my SoftRF device is receiving FLARM signals of other gliders in legacy mode but it is not sending any signals to the gliders. I tested it on different FLARM devices in my gliding club and none of them received the signal of my SoftRF. Do I have to change something in my settings?

## To Reproduce

## Expected behavior
Red light on FLARM device if approaching.

DanD222, 2019-04-03:
This may be correct behaviour of FLARM units, if the testing was done on the ground on stationary units - see the "THRE" parameter in the FLARM Configuration document:
http://asa-soaring.org/documents/FLARMConfiguration.pdf

Also be aware of the "Nearest" and "Collision" FLARM operating modes:
https://flarm.com/wp-content/uploads/man/FLARM_OperatingManual_E.pdf

A good way to test whether a FLARM unit is receiving SoftRF signals is if the FLARM is connected to a gliding computer such as XCSoar via a Bluetooth module such as the Soartronic modules. Signals received by the FLARM units are then displayed on the XCSoar FLARM display regardless of whether the LEDs on the actual FLARM unit are lit or not.

GliderPil0t, 2019-04-03:
Oh thanks for the response. I'll try that soon :)

closed 2019-04-03 by GliderPil0t