lyusupov/SoftRF #56: Question Raspbery Edition

sunlord2k, 2019-04-29:
I am woking on a rasperry based setup for my glider.
But I cannot get the nRF905 woking connected to the raspberry.
The bsm2835 is working fine, with acces to all pins incl the spi pins.

But SoftRF does not find the nRF905.
"WARNING! Neither SX1276, CC13XX or NRF905 RFIC is detected!"

Is the nRF905 implemted correctly already?
Can you post the corect wiring of the nRF905 to the Raspberry, so I can double check?

lyusupov, 2019-04-29:
It was specified [there](https://github.com/lyusupov/SoftRF/wiki/Raspberry-Edition#architecture) that:
<sup>1</sup> - **nRF905 IC driver is not well tested** yet **on Raspberry Pi**.

GPIO mapping for nRF905 IC is "hardwired" into it"s library code:
https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/libraries/nRF905/nRF905_config.h

lyusupov, 2019-05-13:
https://github.com/lyusupov/SoftRF/commit/1b616d3e362c41e9a91a6365fbb0a57d69db6b42

closed 2019-05-13 by lyusupov