lyusupov/SoftRF #57: PCB Sourcefiles

datamann, 2019-05-21:
Hi,

Is it possible to get the source files for the PCB?
I would like to make some modification to it and it would be nice to not have to redraw all from scratch.

Regards
Stig

lyusupov, 2019-05-21:
There are at least 4 different PCB design files in this project: Shield, LoRa, Adapter/UAV, UAT.
Which one?

datamann, 2019-05-22:
Hi lyusupov,

I was thinking of the PCB for the LoRa Module
https://github.com/lyusupov/SoftRF/wiki/SoftRF-LoRa-module

Regards
Stig

lyusupov, 2019-05-22:
Negative.
Your [own project](https://github.com/datamann/LoRaGPS) has nothing related to this one.
To state that it is 'all yours' you have to design your PCB from scratch.
I believe that it is fair enough.

datamann, 2019-05-22:
You're right my project is not related to SoftRf but still your PCB is great work and I will of course share my changes (if I'm able to do the changes. I am a newbie.) with you if you are intrested. I do not recall that I've stated anywhere that is't "all mine", basically nothing is mine, there are always someone who have done all I do before, in some way or another. All I do is to package all the pieces in a different way : https://aconfluence.no/display/SS/LoRaGPS

All I wanted to do was to add a 5v to 3.3v regulator. I burned out one LoRa module because my arduino nano didn't output 3.3v put 3.9v and I didn't checked it before use... bummer...

Well if I have to redraw, I will and I will send you the files anyway so you can deside for your self if you want them or not...

I have printed and bought all SoftRf parts and will build your project later. My project actually started with SoftRF but then I got sidetracked...

Regards
Stig

closed 2019-05-22 by datamann