lyusupov/SoftRF #58: 2 hours SoftRF dhcp lease when used with kobo

bambukouk, 2019-05-22:
When using SoftRF with kobo, wi-fi disconnects exactly after 120 minutes, everytime, on at least two different models of kobo that I have tested.
It is being suggested that it is caused by short dhcp lease on SoftRF? and it could be solved by changing it on SoftRF?
I have seen it, both in flight, and when bench testing.

lyusupov, 2019-05-22:
Report is filed not in compliance with the [template](https://github.com/lyusupov/SoftRF/blob/master/.github/ISSUE_TEMPLATE/softrf-issue-template.md).

bambukouk, 2019-05-22:
I am sorry for getting it wrong.
I will have another go using the template (or should I re-open/edit this one?)

lyusupov, 2019-05-22:
Re-open is fine.

bambukouk, 2019-05-25:
Hardware:
Prime Mark II
[image]
[image]

Firmware version:
[image]

 Firmware settings:
[image]

Describe the bug (or ask a question):  
When using SoftRF with kobo, wi-fi disconnects exactly after 120 minutes, everytime, on at least two different models of kobo that I have tested.
It is being suggested that it is caused by short dhcp lease on SoftRF? and it could be solved by changing it on SoftRF?
By default kobo don't have 'auto renew' of dhcp lease...
I have seen it, both in flight, and when bench testing.

To Reproduce:
Connect any kobo (I tried mini and Clara HD) to SoftRF

Expected behavior:
Wi-fi connection to remain on for at least 5 hrs or (more desirably) for the whole day?
At the moment it disconnects after 120 minutes and the only way to get it going again is to re-start kobo (this would cause the loss of all task info)

bambukouk, 2019-05-30:
> 
> 
> Re-open is fine.

I don't know whether you meant to re-open as a new issue or re-open by adding comment to this one using correct template?
If I am wrong (yet again...) please let me know and I will do the necessary.

lyusupov, 2019-05-31:
[image]

bambukouk, 2019-05-31:
I must be particulraly thick...
Can't see re-open anywhere?
[image]
Sorry

lyusupov, 2019-06-04:
I've assigned this ticket onto you.
Try again.

bambukouk, 2019-06-04:
I am afraid it doesn't make any difference...
the option to re-open is still not available to me 
I am happy to raise a new issue and cross-refer the two? leaving #58 closed and linked to a new one?

lyusupov, 2019-06-05:
1) **Give me a reference on a** [RFC 2131](https://tools.ietf.org/html/rfc2131) **paragraph where it is specified that: 120 minutes lease time is a "bad value"**.

Instead, the standard gives an option for DHCP server (ESP32, in this case) to choose an appropriate value, then any DHCP client (Kobo, in your case) have to comply with.

'120 minutes' lease value is a part of Espressif SDK: https://github.com/espressif/esp-idf/blob/master/components/lwip/include/apps/dhcpserver/dhcpserver.h
and SoftRF has no control over it.
Moreover, I consider that this is a reasonable value, provided that ESP32 AP can serve very limited number of Wi-Fi clients. Greater lease time value may cause exhaust of spare IP addresses pretty soon, when DHCP clients do abrupt disconnect/reconnect sequence.

So, if you have your own strong arguments against 120 minutes - argue with Espressif SDK developers, not me.

2) **Give me a reference on a paragraph in my SoftRF GitHub page(s) where I say that Kobo is a good, approved client for Wi-Fi connection to a SoftRF unit**.

By choosing a device, not mentioned as known good one, you are taking all the associated risks on your own.

Friendly reminder that **SoftRF firmware comes free of charge** and without even the implied warranty of FITNESS FOR A PARTICULAR PURPOSE.  See the GNU General Public License for more details.

**Ask customer service of a product you've paid for to make you feel happy**.

So, if you are not satisfied with your TTGO T-Beam **hardware** (you should had to read this [Disclamer](https://github.com/lyusupov/SoftRF/wiki/Prime-Edition-MkII#disclaimer) before purchasing the board ) - work with your board seller.
If you are not satisfied with your Kobo compliance to the RFC 2131 DHCP standard - work with Rakuten customer service. 

**Resolution**: invalid.

closed 2019-06-05 by lyusupov

bambukouk, 2019-06-05:
Thank you for detailed explanation and your help.

TexasUK, 2019-06-05:
Yes thanks for the info.

I've asked espressif about extending the lease time, but in the meantime,
could we change the option in the code you pointed to and compile the code,
then flash the chip on the board?

Is that even possible?

Cheers,
Daren