lyusupov/SoftRF #60: Raspberry Pi 3 with Dragino Lora/GPS HAT

pat-fly, 2019-06-04:
**ATTENTION** ! 

- Use this form to report bugs in released SoftRF firmware, hardware and web pages only ;
- Prior to asking a question, look into ["Closed issues"](https://github.com/lyusupov/SoftRF/issues?q=is%3Aissue+is%3Aclosed) area to see if it was answered before ;
- No any "ideas" or new feature requests, please ! New feature requests do not subject for taking into consideration unless they come from co-developers ;
- "Enhancements" of existing features are allowed only when you are ready to implement them by yourself ;

Recommended:
- screenshot or photo width - from 600 to 800 pixels ;
- aspect ratio - 4:3, 3:4, 16:9 or 9:16.
- substitute 'space' by 'x' character to mark a hardware checkbox

--------------------

## Hardware

Raspberry Pi 3
Dragino Lora/GPS HAT

## Firmware version

SoftRF-v1.0-rc6-RPi

## Describe the bug (or ask a question)
I checked all the wirings and pins in Platform_RPi.h and it should be correct. But still I get this error:

```
pi@raspberrypi:~/SoftRF $ sudo ./SoftRF
WARNING! Neither SX1276, CC13XX or NRF905 RFIC is detected!
```

On serial port /dev/serial0 I get the GPS NMEA output. So this works fine.

lyusupov, 2019-06-04:
1) photo of your hardware (**required**) ?

2) test your LoRa board with some other software (say,  provided by manufacturer) to ensure that it is not defective.

pat-fly, 2019-06-04:
So photo of the hardware are here:
I'm now on the RPI Zero W again. But basically it is the same as for the RPI 3.
 
[image]
[image]

 It is the v1.3 of Dragino Lora/GPS HAT and for use with sx1276 Library I need to connect these two wires as written in the wiki of the Dragino project.

I can initiate the single_chan_pkt_fwd for example and read the registers of the SX1276:

```
pi@raspberrypi:~/lora/single_chan_pkt_fwd $ sudo ./single_chan_pkt_fwd 
SX1276 detected, starting.
0x01:	 0x85
0x06:	 0xD9
0x07:	 0x20
0x08:	 0x00
0x09:	 0x4F
0x0A:	 0x09
0x0B:	 0x2B
0x0C:	 0x23
0x0D:	 0x00
0x0E:	 0x80
0x0F:	 0x00
0x10:	 0x00
0x11:	 0x00
0x12:	 0x00
0x13:	 0x00
0x14:	 0x00
0x15:	 0x00
0x16:	 0x00
0x17:	 0x00
0x18:	 0x04
0x19:	 0x00
0x1A:	 0x00
0x1B:	 0x25
0x1C:	 0x00
0x1D:	 0x72
0x1E:	 0xC4
0x1F:	 0x05
0x20:	 0x00
0x21:	 0x08
0x22:	 0x40
0x23:	 0x80
0x24:	 0xFF
0x25:	 0x00
0x26:	 0x0C
0x27:	 0x00
Gateway ID: 00:00:00:ff:ff:00:00:00
Listening at SF12 on 868.500000 Mhz.
------------------
stat update: {"stat":{"time":"2019-06-04 11:20:12 GMT","lati":0.00000,"long":0.00000,"alti":0,"rxnb":0,"rxok":0,"rxfw":0,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"Single Channel Gateway","mail":"","desc":""}}
```

lyusupov, 2019-06-04:
I have Zero and 2B. Both are doing fine to me.
I don't have W or 3 and have no plans to purchase them.

Are you able to set up an SSH tunnel to your Pi 3 so I could temporarily access this board remotely ?

Otherwise I see no way to help you - so you will have to be on your own.

When I can not see an issue on the hardware I have -  I typically close that ticket with 'unable to reproduce' resolution. Unless originator of the issue is not ready to process that ticket by himself.

pat-fly, 2019-06-04:
ok thanks. I am now trying with a 2B from scratch.
When compiling i.e. make -f Makefile.RPi i get one warning. I don't thing this is relevant but just in case:
```
config.status: executing depfiles commands
make[1]: Verzeichnis „/home/pi/SoftRF/software/firmware/source/libraries/bcm2835“ wird betreten
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/bash /home/pi/SoftRF/software/firmware/source/libraries/bcm2835/missing aclocal-1.15 -I m4
aclocal-1.15: warning: couldn't open directory 'm4': No such file or directory
 cd . && /bin/bash /home/pi/SoftRF/software/firmware/source/libraries/bcm2835/missing automake-1.15 --gnu
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/bash /home/pi/SoftRF/software/firmware/source/libraries/bcm2835/missing autoconf
/bin/bash ./config.status --recheck
```

If this does not work now, I would be happy if you can try to help me. Is there a possibility to send you the IP and temporary password in a private message?

lyusupov, 2019-06-04:
I am not certain if GitHub has PMs.
Use e-mail address specified in SoftRF.ino
Expect to keep it running for 24 hours or until I'll say that I no longer need it, whatever comes first. 

TCP 22 port forwarding in a (Wi-Fi) router from WAN's IP address onto LAN Pi's address is typically sufficient to establish this remote access.

lyusupov, 2019-06-04:
**I don't think** that this:

[image]

is a good idea

pat-fly, 2019-06-04:
this was needed for a library I used that is linked on the dragino wiki. Obviously they coded the wrong connection. It links GPIO 8 to 25. But I will remove it and test again. If you are using GPIO 25 for something else, then this could be a cause of the problem.