lyusupov/SoftRF #61: Flarm and SoftRF not seeig each other

sixpack-de, 2019-06-18:
- [X ]   Prime Mark II
- [ ]   Standalone
- [ ]   UAV
- [ ]   UAT module
- [ ]   SkyView

[image]

## Firmware version
[image]

## Firmware settings

[image]

## Describe the bug (or ask a question)
Probably its a setting I am to dumb to set. 

System is stating to send out Flarm but cannot be recieved by Flarm systems (tested at my glider port and recreated with a dedicated Flarm Setup, see above). 

System is not recieving (stays at 0) also tested at airport and recreated at home 

Please be not confused with the skyview I was testing something when the pic was done, the ipad screen is the Mark II live.

## Expected behavior
I was expecting to see FLARM equipped gliders (and be seen).

Cool project BTW!!!

Ben64Smith, 2019-06-18:
> small](https://user-images.githubusercontent.com/51480811/59678122-d9349000-91cb-11e9-9610-1a00db7f261c.JPG)

Change protocol from OGNTP to LEGACY then it will se and be seen by FLARM.

sixpack-de, 2019-06-18:
I will test tonight ... THX!!

lyusupov, 2019-06-18:
https://github.com/lyusupov/SoftRF/wiki/Settings#protocol-1

**Resolution:** invalid

closed 2019-06-18 by lyusupov