lyusupov/SoftRF #71: SoftRF on Lora32 Oled

mbatini, 2019-10-23:
**ATTENTION** ! 

- Use this form to report bugs in released SoftRF firmware, hardware and web pages only ;
Not a bug

- Prior to asking a question, look into ["Closed issues"](https://github.com/lyusupov/SoftRF/issues?q=is%3Aissue+is%3Aclosed) area to see if it was answered before ;
Looked into but couldn't find out info

- No any "ideas" or new feature requests, please ! New feature requests do not subject for taking into consideration unless they come from co-developers ;
- "Enhancements" of existing features are allowed only when you are ready to implement them by yourself ;

Recommended:
- screenshot or photo width - from 600 to 800 pixels ;
- aspect ratio - 4:3, 3:4, 16:9 or 9:16.
- substitute 'space' by 'x' character to mark a hardware checkbox

--------------------

## Hardware

- [ ]   Prime Mark II
- [ ]   Standalone
- [ ]   UAV
- [ ]   Raspberry Edition
- [ ]   UAT module
- [ ]   SkyView
- [ ]   Flight Recorder

Wemos Lora32 Oled

< photo of your hardware > (**required**)

## Firmware version

<  screenshot of WebUI status page > (**required**)

## Firmware settings

<  screenshot of WebUI settings page > (**required**)

## Describe the bug (or ask a question)
SoftRF repo cloned and compiled and flashed
Is it possible to use a neo 6m gps module ? Looking inside code for pin definition, but need hints. 

I see that connecting GPS RX to board PIN 16 (OLED RST) Oled shows up. But would like any hint.

TIA
[image]
[image]

## To Reproduce

## Expected behavior

lyusupov, 2019-10-24:
This board [does not perform good enough](https://github.com/lyusupov/SoftRF/releases/tag/1.0-rc2) to qualify it for SoftRF purpose.

**Relosution**: invalid

closed 2019-10-24 by lyusupov

nbonniere, 2019-10-31:
Seems like a reasonable board, with SD card interface (could record an IGC file), and needs an external GPS (simple enough to connect), but has no PSRAM. Is the lack of PSRAM why it is not good ?

nbonniere, 2019-10-31:
I guess there are two versions. V2.1 seems OK.