lyusupov/SoftRF #74: Wrong OGNTP packets from SoftRg RaspberryPi edition on RELAY from dump1090

acasadoalonso, 2020-01-04:
**ATTENTION** ! 

- Use this form to report bugs in released SoftRF firmware, hardware and web pages only ;
- Prior to asking a question, look into ["Closed issues"](https://github.com/lyusupov/SoftRF/issues?q=is%3Aissue+is%3Aclosed) area to see if it was answered before ;
- No any "ideas" or new feature requests, please ! New feature requests do not subject for taking into consideration unless they come from co-developers ;
- "Enhancements" of existing features are allowed only when you are ready to implement them by yourself ;

Recommended:
- screenshot or photo width - from 600 to 800 pixels ;
- aspect ratio - 4:3, 3:4, 16:9 or 9:16.
- substitute 'space' by 'x' character to mark a hardware checkbox

--------------------

## Hardware

- [ ]   Standalone
- [ ]   Prime Mark II
- [ ]   Dongle
- [ ]   SkyView EZ
- [ x]   Raspberry Edition
- [ ]   UAV
- [ ]   UAT978 module
- [ ]   Flight Recorder

< photo of your hardware > (**required**)

Same that previous issue.

## Firmware version
Same than previous issue

SoftRF-RPi FW.REV: 1.0-rc7 DEV.ID: 7F0101
SX1276 RFIC is detected.

## Firmware settings

{ echo "{class:SOFTRF,mode:RELAY}" ; cat /dev/ttyAMA0 ; } | sudo ./SoftRF 

## Describe the bug (or ask a question)

After solving some installation issues, now the dragino Hat that is properly installed, the GPS works and the dump1090 is working properly

However the RELAY packets from dump1090 in OGNTP format seems to be wrong:
This is the output of the telnet localhost 50001 of an OGN station (LEMD) nearby the RPi+dragino HAT with the SoftRF SW+dump1090
----
0.404sec:868.393MHz:  s9:0:000000 000142: [ -65.08855,-181.61252]deg  790m  +0.5m/s 162.6m/s 054.1deg  +5.3deg/sec O:9/0 11x17m 01_o -6.78kHz 45.8/61.5dB/0  0e 11734.5km 224.2deg  +2.3deg           B0885
0.423sec:868.393MHz:  s9:0:000000 000142: [ -65.08855,-181.61252]deg  790m  +0.5m/s 162.6m/s 054.1deg  +5.3deg/sec O:9/0 11x17m 01_o -6.79kHz 46.0/61.5dB/0  0e 11734.5km 224.2deg  +2.3deg           B0885
APRS <- RND000000>OGNTRK,qOR:/000142h6505.31S/18136.75W^054/316/A=002592 !W31! idA4000000 +099fpm +1.8rot FL029.04 46.0dB 0e -6.8kHz gps11x17
HTTP_Server.Exec() ... Client from 127.0.0.1:59660
HTTP_Server.Exec() ... Request for /
0.804sec:868.393MHz:  s9:0:000000 000142: [ -65.08855,-181.61252]deg  790m  +0.5m/s 162.6m/s 054.1deg  +5.3deg/sec O:9/0 11x17m 01f_ -6.79kHz 46.2/62.0dB/0  0e 11734.5km 224.2deg  +2.3deg           B0885
0.505sec:868.393MHz:  s9:0:000000 000142: [ -65.08855,-181.61252]deg  790m  +0.5m/s 162.6m/s 054.1deg  +5.3deg/sec O:9/0 11x17m 01_o -6.79kHz 46.5/62.0dB/0  0e 11734.5km 224.2deg  +2.3deg           B0885
APRS -> # aprsc 2.1.4-g408ed49 4 Jan 2020 00:01:59 GMT GLIDERN2 37.187.244.41:14580
APRS time - system time = +0
0.452sec:868.393MHz:  s9:0:000000 000142: [ -65.08855,-181.61252]deg  790m  +0.5m/s 162.6m/s 054.1deg  +5.3deg/sec O:9/0 11x17m 01_o -6.80kHz 46.8/62.5dB/0  0e 11734.5km 224.2deg  +2.3deg           B0885
------

The coordinates are wrong:
-65.088 and -181.61 are totally wrong.

The traffic from dump190 over port 8080 is OK
[
{"hex":"34558e", "squawk":"3765", "flight":"AEA057  ", "lat":40.512876, "lon":-4.071496, "validposition":1, "altitude":13800,  "vert_rate":1664,"track":243, "validtrack":1,"speed":386, "messages":294, "seen":286}
]
## To Reproduce

## Expected behavior
It looks that it could be a problem with the baud rate to the LoRa perhaps ???

lyusupov, 2020-01-04:
Try out https://github.com/lyusupov/SoftRF/commit/adb691f6a42c8f02b776b602fc3984bba4d7fc43

Think about to enable some debug [here](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/Platform_RPi.cpp#L613) and [there](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/Platform_RPi.cpp#L631), then capture an output log.

Be aware that:
* ADS-B traffic outside [this area](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/Platform_RPi.cpp#L624) will not be relayed and ;
* rebroadcast rate [is limited](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/Platform_RPi.cpp#L629) by legitimate duty cycle limit for ISM band .

acasadoalonso, 2020-01-04:
With the patch adb691f do not transmit anything
without the patch and the debug flags as 1 same behaviours as expected:
0.359sec:868.393MHz:  s9:0:000000 121342: [ -65.08855,-181.61252]deg  790m  +0.5m/s 162.6m/s 054.1deg  +5.3deg/sec O:9/0 11x17m 01_o -7.06kHz 48.8/64.5dB/0  0e 11734.5km 224.2deg  +2.3deg           B0885
APRS <- RND000000>OGNTRK,qOR:/121242h6505.31S/18136.75W^054/316/A=002592 !W31! idA4000000 +099fpm +1.8rot FL029.04 50.5dB 0e -7.3kHz gps11x17
0.484sec:868.393MHz:  s9:0:000000 121342: [ -65.08855,-181.61252]deg  790m  +0.5m/s 162.6m/s 054.1deg  +5.3deg/sec O:9/0 11x17m 01_o -7.04kHz 50.5/66.0dB/0  0e 11734.5km 224.2deg  +2.3deg           B0885
APRS <- RND000000>OGNTRK,qOR:/121342h6505.31S/18136.75W^054/316/A=002592 !W31! idA4000000 +099fpm +1.8rot FL029.04 50.5dB 0e -7.0kHz gps11x17

and the debug output log as ZERO !!!

SoftRF-RPi FW.REV: 1.0-rc7 DEV.ID: 7F0101
SX1276 RFIC is detected.
0000000000000000000000000000000000000000
$GPGLL,,,,,121849.084,V,N*71
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,1,1,01,15,,,17*7A
$GPRMC,121849.084,V,,,,,0.00,0.00,040120,,,N*41
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,121849.084,,,,,0,0,,,M,,M,,*43
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,121849.084,,,,,0,0,,,M,,M,,*43
$GPGLL,,,,,121849.084,V,N*71
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,1,1,01,15,,,17*7A
$GPRMC,121849.084,V,,,,,0.00,0.00,040120,,,N*41
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,121849.084,,,,,0,0,,,M,,M,,*43
$GPGLL,,,,,121849.084,V,N*71
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,1,1,01,15,,,17*7A
0000000000000000000000000000000000000000
$GPRMC,121849.084,V,,,,,0.00,0.00,040120,,,N*41
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,121849.084,,,,,0,0,,,M,,M,,*43
$GPGLL,,,,,121849.084,V,N*71
$GPGGA,121850.084,,,,,0,0,,,M,,M,,*4B
$GPGLL,,,,,121850.084,V,N*79
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,1,1,01,15,,,17*7A
$GPRMC,121850.084,V,,,,,0.00,0.00,040120,,,N*49
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGGA,121849.084,,,,,0,0,,,M,,M,,*43
$GPGLL,,,,,121849.084,V,N*71
$GPGGA,121850.084,,,,,0,0,,,M,,M,,*4B
$GPGLL,,,,,121850.084,V,N*79
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,1,1,01,15,,,17*7A
0000000000000000000000000000000000000000

follow after a wile by 👍 
0000000000000000000000000000000000000000
$GPGLL,,,,,121939.084,V,N*77
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,1,1,01,15,,,15*78
$GPRMC,121939.084,V,,,,,0.00,0.00,040120,,,N*47
$GPVTG,0.00,T,,M,0.00,N,0.00,K,N*32
$GPGSV,1,1,01,15,,,14*79
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000
0000000000000000000000000000000000000000

I started to suspect that my dragino board is a faulty one.
I bought it from Aliexpress ZucZug store 
https://www.aliexpress.com/item/32860051864.html?spm=a2g0s.9042311.0.0.39dd4c4dYoZsF7
and do not mention the Dragino  version on the silkscreen.

I ordered a new one from the Aliexpress Dragino store now:
https://www.aliexpress.com/item/32891200256.html?spm=a2g0s.9042311.0.0.39dd4c4dYoZsF7
that it mentions the Dragino version 1.4 on the silkscreen, it is more expensive, but I hope that it will be better.

acasadoalonso, 2020-01-04:
Now with the GPS lock !!!

---------------------------

0000000000000000000000000000000000000000
$GPGGA,134501.000,4030.4257,N,00341.7276,W,1,4,1.84,-53.6,M,51.7,M,,*5A
$GPGLL,4030.4257,N,00341.7276,W,134501.000,A,A*48
$GPGSA,A,3,24,05,15,28,,,,,,,,,5.10,1.84,4.76*03
$GPGSV,3,1,11,13,72,058,17,15,64,314,17,28,46,064,18,05,33,181,16*78
$GPGSV,3,2,11,24,30,260,18,30,19,061,,20,16,317,,17,09,109,13*79
$GPGSV,3,3,11,21,05,289,,19,01,130,,12,01,200,*47
$GPRMC,134501.000,A,4030.4257,N,00341.7276,W,1.17,359.18,040120,,,A*79
$GPVTG,359.18,T,,M,1.17,N,2.16,K,A*39
$GPGGA,134502.000,4030.4254,N,00341.7286,W,1,4,1.84,-53.8,M,51.7,M,,*5B
$GPGLL,4030.4254,N,00341.7286,W,134502.000,A,A*47
$GPGSA,A,3,24,05,15,28,,,,,,,,,5.10,1.84,4.76*03
0000000000000000000000000000000000000000
$GPGSV,3,1,11,13,72,058,17,15,64,314,17,28,46,064,18,05,33,181,15*7B
$GPGSV,3,2,11,24,30,260,19,30,19,061,,20,16,317,,17,09,109,12*79
$GPGSV,3,3,11,21,05,289,,19,01,130,,12,01,200,*47
$GPRMC,134502.000,A,4030.4254,N,00341.7286,W,1.19,9.72,040120,,,A*72
$GPVTG,9.72,T,,M,1.19,N,2.21,K,A*39
0000000000000000000000000000000000000000
$GPGGA,134503.000,4030.4254,N,00341.7296,W,1,4,1.84,-54.0,M,51.7,M,,*54
$GPGLL,4030.4254,N,00341.7296,W,134503.000,A,A*47
$GPGSA,A,3,24,05,15,28,,,,,,,,,5.10,1.84,4.76*03
$GPGSV,3,1,11,13,72,058,17,15,64,314,17,28,46,064,18,05,33,181,14*7A
$GPGSV,3,2,11,24,30,260,19,30,19,061,,20,16,317,,17,09,109,12*79
$GPGSV,3,3,11,21,05,289,,19,01,130,,12,01,200,*47
$GPRMC,134503.000,A,4030.4254,N,00341.7296,W,1.21,6.71,040120,,,A*75
$GPVTG,6.71,T,,M,1.21,N,2.24,K,A*3B

lyusupov, 2020-01-04:
1. **double check** that you've applied the patch.
2. inspect  the Plat...RPi.cpp manually
3. do _make -f Makefile.RPi clean ; make -f Makefile.RPi_

lyusupov, 2020-01-04:
BTW, for your aircraft location example above (taken from JSON) and location of ground station (taken from NMEA):

[image]

Distance between these two points is well above 2 \* ALARM_ZONE_NONE = 20 km.
So the traffic will be filtered out and thus will not re-broadcasted.

acasadoalonso, 2020-01-04:
Yes I updated the code with your patch and recompile the program, no output whatsoever, so I removed the patch. I reverted the patch back just for the test 
I will apply the patch again and recompile 

Btw: on my dump1090 version the aircraft in JSON format is on
http://localhost:8080/data.json 
as shown above and not /data/aircraft.json as mentioned by you and on the wiki pages. I wonder if I am using the latest version? Ver : 1.10.3010.14 ... if not where I can find the latest version ?

I will increase the ALARM zone, just for the testing.
Thanks for your great help.

acasadoalonso, 2020-01-04:
After applying the patch again and increasing the ALARM_ZONE_NONE to 180 kms, this is the output and belo you can see the JSON data as well:
————

SoftRF-RPi FW.REV: 1.0-rc7 DEV.ID: 7F0101
SX1276 RFIC is detected.
$GPGLL,4030.2454,N,00341.9448,W,184046.000,A,A*43
$GPGSA,A,3,25,32,12,14,29,31,,,,,,,1.66,1.36,0.95*02
$GPGSV,3,1,09,25,72,345,13,29,53,190,23,32,46,259,24,12,43,056,18*71
$GPGSV,3,2,09,14,38,297,18,24,30,128,,02,24,072,,31,23,308,19*75
$GPGSV,3,3,09,06,09,039,*45
$GPRMC,184046.000,A,4030.2454,N,00341.9448,W,0.59,234.59,040120,,,A*76
$GPVTG,234.59,T,,M,0.59,N,1.09,K,A*30
$GPRMC,134450.000,A,4030.4239,N,00341.7221,W,0.84,50.74,040120,,,A*4D
$GPVTG,50.74,T,,M,0.84,N,1.55,K,A*06
$GPGGA,134451.000,4030.4245,N,00341.7224,W,1,4,1.47,-52.0,M,51.7,M,,*52
$GPGLL,4030.2455,N,00341.9447,W,184047.000,A,A*4C
$GPGSA,A,3,25,32,12,14,29,31,,,,,,,1.66,1.36,0.95*02
$GPGSV,3,1,09,25,72,345,14,29,53,190,23,32,46,259,24,12,43,056,18*76
$GPGSV,3,2,09,14,38,297,18,24,30,128,,02,24,072,,31,23,308,19*75
$GPGSV,3,3,09,06,09,039,*45
$GPRMC,184047.000,A,4030.2455,N,00341.9447,W,0.17,234.59,040120,,,A*73
$GPVTG,234.59,T,,M,0.17,N,0.31,K,A*30
$GPGSA,A,3,24,05,15,28,,,,,,,,,1.76,1.47,0.98*0C
$GPGSV,3,1,11,13,72,057,18,15,64,314,16,28,46,064,18,05,33,181,17*78
$GPGSV,3,2,11,24,30,260,18,30,19,061,,20,16,317,,17,08,109,14*7F
$GPGSV,3,3,11,21,05,289,,19,01,131,,12,01,200,*46
$GPRMC,134451.000,A,4030.4245,N,00341.7224,W,1.00,32.36,040120,,,A*4D
$GPVTG,32.36,T,,M,1.00,N,1.86,K,A*07
$GPGGA,134452.000,4030.4245,N,00341.7224,W,1,4,1.47,-52.1,M,51.7,M,,*50
$GPGLL,4030.2458,N,00341.9441,W,184048.000,A,A*48
$GPGSA,A,3,25,32,12,14,29,31,,,,,,,1.66,1.36,0.95*02
$GPGSV,3,1,09,25,72,345,14,29,53,190,23,32,46,259,24,12,43,056,18*76
$GPGSV,3,2,09,14,38,297,18,24,30,128,,02,24,072,,31,23,308,19*75
$GPGSV,3,3,09,06,09,039,*45
$GPRMC,184048.000,A,4030.2458,N,00341.9441,W,0.35,234.59,040120,,,A*77
$GPVTG,234.59,T,,M,0.35,N,0.64,K,A*30
$GPGSA,A,3,24,05,15,28,,,,,,,,,1.76,1.47,0.98*0C
$GPGGA,184047.000,4030.2455,N,00341.9447,W,1,6,1.36,699.9,M,51.7,M,,*47
$GPGLL,4030.2460,N,00341.9437,W,184049.000,A,A*43
$GPGSA,A,3,25,32,12,14,29,31,,,,,,,3.47,1.55,3.10*08
$GPGSV,3,1,09,25,72,345,13,29,53,190,23,32,46,259,24,12,43,056,18*71
$GPGSV,3,2,09,14,38,297,18,24,30,128,,02,24,072,,31,23,308,20*7F
$GPGSV,3,3,09,06,09,039,*45
$GPRMC,184049.000,A,4030.2460,N,00341.9437,W,0.27,234.59,040120,,,A*7F
$GPVTG,234.59,T,,M,0.27,N,0.50,K,A*34
$GPGGA,134452.000,4030.4245,N,00341.7224,W,1,4,1.47,-52.1,M,51.7,M,,*50
$GPGLL,4030.2461,N,00341.9432,W,184050.000,A,A*4F
$GPGSA,A,3,25,32,12,14,29,31,,,,,,,1.66,1.36,0.95*02
$GPGSV,3,1,09,25,72,345,14,29,53,190,24,32,46,259,24,12,43,056,17*7E
$GPGSV,3,2,09,14,38,297,18,24,30,128,,02,24,072,,31,23,308,20*7F
$GPGSV,3,3,09,06,09,039,*45
$GPRMC,184050.000,A,4030.2461,N,00341.9432,W,0.29,234.59,040120,,,A*7D
$GPVTG,234.59,T,,M,0.29,N,0.54,K,A*3E
$GPGGA,184051.000,4030.2461,N,00341.9433,W,1,6,1.36,699.6,M,51.7,M,,*4B
$GPGLL,4030.2461,N,00341.9433,W,184051.000,A,A*4F
$GPGSA,A,3,25,32,12,14,29,31,,,,,,,1.66,1.36,0.95*02
$GPGSV,3,1,09,25,72,345,13,29,53,190,24,32,46,259,24,12,43,056,17*79
$GPGSV,3,2,09,14,38,297,18,24,30,128,,02,24,072,,31,23,308,20*7F
$GPGSV,3,3,09,06,09,039,*45
$GPRMC,184051.000,A,4030.2461,N,00341.9433,W,0.26,234.59,040120,,,A*72
$GPVTG,234.59,T,,M,0.26,N,0.48,K,A*3C
$GPGGA,184052.000,4030.2462,N,00341.9433,W,1,6,1.36,699.6,M,51.7,M,,*4B
$GPGLL,4030.2462,N,00341.9433,W,184052.000,A,A*4F
$GPGSA,A,3,25,32,12,14,29,31,,,,,,,1.66,1.36,0.95*02
$GPGSV,3,1,09,25,72,345,15,29,53,190,24,32,46,259,24,12,43,056,17*7F
$GPGSV,3,2,09,14,38,297,18,24,30,128,,02,24,072,,31,23,308,20*7F
$GPGSV,3,3,09,06,09,039,*45
$GPRMC,184052.000,A,4030.2462,N,00341.9433,W,0.45,234.59,040120,,,A*77
$GPVTG,234.59,T,,M,0.45,N,0.84,K,A*39
$GPGGA,184053.000,4030.2463,N,00341.9433,W,1,6,1.55,699.6,M,51.7,M,,*4E
[... 299 lines cut ...]
{"hex":"4ca7b5", "squawk":"0502", "flight":"RYR598Y ", "lat":40.794525, "lon":-4.247681, "validposition":1, "altitude":37000,  "vert_rate":0,"track":185, "validtrack":1,"speed":439, "messages":341, "seen":0},
{"hex":"020118", "squawk":"0000", "flight":"", "lat":0.000000, "lon":0.000000, "validposition":0, "altitude":0,  "vert_rate":0,"track":0, "validtrack":0,"speed":0, "messages":2, "seen":67},
{"hex":"3964ea", "squawk":"6436", "flight":"", "lat":39.776699, "lon":-4.406250, "validposition":1, "altitude":38000,  "vert_rate":0,"track":23, "validtrack":1,"speed":433, "messages":213, "seen":0},
{"hex":"3964f1", "squawk":"6463", "flight":"", "lat":39.611801, "lon":-4.499319, "validposition":1, "altitude":34000,  "vert_rate":0,"track":23, "validtrack":1,"speed":419, "messages":50, "seen":1},
{"hex":"344217", "squawk":"5046", "flight":"AEA87XB ", "lat":40.659668, "lon":-3.464600, "validposition":1, "altitude":7725,  "vert_rate":3584,"track":40, "validtrack":1,"speed":253, "messages":540, "seen":0},
{"hex":"406ae3", "squawk":"2247", "flight":"BAW464  ", "lat":40.609222, "lon":-4.224060, "validposition":1, "altitude":15750,  "vert_rate":-1280,"track":185, "validtrack":1,"speed":359, "messages":548, "seen":1},
{"hex":"345559", "squawk":"0000", "flight":"", "lat":0.000000, "lon":0.000000, "validposition":0, "altitude":3950,  "vert_rate":0,"track":0, "validtrack":0,"speed":0, "messages":8, "seen":123},
{"hex":"344487", "squawk":"4662", "flight":"", "lat":40.574844, "lon":-3.303894, "validposition":1, "altitude":34000,  "vert_rate":0,"track":8, "validtrack":1,"speed":420, "messages":165, "seen":0},
[... cut ...]
{"hex":"39cea0", "squawk":"4466", "flight":"TVF86DA ", "lat":40.728745, "lon":-3.854187, "validposition":1, "altitude":36000,  "vert_rate":0,"track":23, "validtrack":1,"speed":409, "messages":5454, "seen":73},
{"hex":"3c56f6", "squawk":"4711", "flight":"EWG7RR  ", "lat":40.471724, "lon":-2.972051, "validposition":1, "altitude":38000,  "vert_rate":0,"track":39, "validtrack":1,"speed":439, "messages":1007, "seen":114},
{"hex":"4caf7e", "squawk":"0750", "flight":"SAS6205 ", "lat":39.307406, "lon":-4.292725, "validposition":1, "altitude":39000,  "vert_rate":64,"track":210, "validtrack":1,"speed":452, "messages":4191, "seen":221}
]
pi@SoftRF:~ $

acasadoalonso, 2020-01-04:
It seems based on the documentation that the .json format is different and I need to install the dump1090 from the Pia ware website
https://github.com/flightaware/dump1090/blob/master/README-json.md

acasadoalonso, 2020-01-04:
https://github.com/flightaware/dump1090

lyusupov, 2020-01-04:
You are likely using very, very old dump1090.

This is how json file should to look like:

```
{ "now" : 1542618363.7,
  "messages" : 3406,
  "aircraft" : [
    {"hex":"4d010c","messages":2,"seen":11.7,"rssi":-35.5},

< ... skipped .. >

    {"hex":"424591","squawk":"4152","flight":"AFL1247 ","lat":56.126823,"lon":37                                                                             .311287,"nucp":7,"seen_pos":1.1,"altitude":4225,"vert_rate":-640,"track":238,"sp                                                                             eed":249,"messages":613,"seen":0.0,"rssi":-34.9}
  ]
}
```

acasadoalonso, 2020-01-04:
I will install the flightaware version, it looks as you described !!!
Thanks

lyusupov, 2020-01-04:
JSON input should have 'aircraft' token being defined.

SoftRF's JSON parser will not do the job otherwise:
https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/JSONHelper.cpp#L536

lyusupov, 2020-01-04:
My dump1090 is actually not most recent. one It was built from 'mutability' GitHub repo before they've migrated onto 'flightaware' branch:

```
pi@raspberrypi:~ $ which dump1090
/usr/local/bin/dump1090
pi@raspberrypi:~ $ ls -l /usr/local/bin/dump1090
-rwxr-xr-x 1 root staff 319052 Dec 22  2017 /usr/local/bin/dump1090
pi@raspberrypi:~ $ dump1090 --help
-----------------------------------------------------------------------------
| dump1090 ModeS Receiver                              dump1090-mutability  |
-----------------------------------------------------------------------------
--device-index <index>   Select RTL device (default: 0)
--gain <db>              Set gain (default: max gain. Use -10 for auto-gain)
--enable-agc             Enable the Automatic Gain Control (default: off)
--freq <hz>              Set frequency (default: 1090 Mhz)
--ifile <filename>       Read data from file (use '-' for stdin)
--interactive            Interactive mode refreshing data on screen
--interactive-rows <num> Max number of rows in interactive mode (default: 15)
--interactive-ttl <sec>  Remove from list if idle for <sec> (default: 60)
--interactive-rtl1090    Display flight table in RTL1090 format
--raw                    Show only messages hex values

.....

```

I suppose that current 'flightaware' variant should be fine for SoftRF...

acasadoalonso, 2020-01-04:
I will test it with the FlighAware version and report the outcome 

AC/.

acasadoalonso, 2020-01-04:
This the aircraft.json file from the dump1090 version by FlightAware

VERSION:

| dump1090 ModeS Receiver                                 dump1090-fa 3.8.0 |
| build options: ENABLE_RTLSDR ENABLE_BLADERF SC16Q11_TABLE_BITS=8          

Output to a file:

{ "now" : 1578177576.2,
  "messages" : 82564,
  "aircraft" : [
    {"hex":"40768d","flight":"EZY74EL ","alt_baro":35000,"alt_geom":35875,"gs":450.4,"ias":267,"tas":448,"mach":0.788,"track":184.5,"track_rate":0.00,"roll":-0.2,"mag_heading":178.8,"baro_rate":0,"geom_rate":-32,"squawk":"0530","emergency":"none","category":"A3","nav_qnh":1012.8,"nav_altitude_mcp":35008,"lat":40.474365,"lon":-4.403931,"nic":8,"rc":186,"seen_pos":0.4,"version":2,"nic_baro":1,"nac_p":9,"nac_v":1,"sil":3,"sil_type":"perhour","gva":2,"sda":3,"mlat":[],"tisb":[],"messages":2508,"seen":0.1,"rssi":-16.0},
    {"hex":"4ca75e","category":"A3","version":2,"sil_type":"perhour","mlat":[],"tisb":[],"messages":3504,"seen":72.9,"rssi":-25.6},
    {"hex":"0d0bba","alt_baro":18775,"alt_geom":19825,"gs":473.6,"ias":337,"tas":442,"mach":0.712,"track":253.1,"track_rate":0.03,"roll":0.4,"mag_heading":254.7,"baro_rate":1472,"geom_rate":1536,"squawk":"2645","category":"A5","lat":40.513504,"lon":-4.559265,"nic":8,"rc":186,"seen_pos":50.7,"version":2,"nic_baro":1,"nac_p":9,"nac_v":1,"sil":3,"sil_type":"perhour","gva":2,"sda":2,"mlat":[],"tisb":[],"messages":2672,"seen":30.4,"rssi":-24.6},
    {"hex":"4bb152","category":"A5","version":2,"sil_type":"perhour","mlat":[],"tisb":[],"messages":3139,"seen":99.3,"rssi":-26.6},
    {"hex":"3c0d09","alt_baro":38000,"alt_geom":38750,"gs":411.5,"ias":248,"tas":442,"mach":0.784,"track":40.3,"track_rate":0.00,"roll":0.2,"mag_heading":42.7,"baro_rate":0,"geom_rate":32,"squawk":"6257","category":"A3","nav_qnh":1013.6,"nav_altitude_mcp":38016,"nav_altitude_fms":38000,"nav_heading":44.3,"lat":40.588806,"lon":-2.763306,"nic":8,"rc":186,"seen_pos":28.2,"version":2,"nic_baro":1,"nac_p":9,"nac_v":1,"sil":3,"sil_type":"perhour","mlat":[],"tisb":[],"messages":3643,"seen":28.2,"rssi":-24.9},
    {"hex":"3455d9","flight":"VOE2725 ","alt_baro":38000,"alt_geom":38825,"gs":409.6,"ias":243,"tas":434,"mach":0.768,"track":41.2,"track_rate":0.00,"roll":0.0,"mag_heading":45.4,"baro_rate":0,"geom_rate":0,"squawk":"6265","emergency":"none","category":"A3","nav_qnh":1012.8,"nav_altitude_mcp":38016,"lat":40.801454,"lon":-2.593273,"nic":7,"rc":371,"seen_pos":0.2,"version":2,"nic_baro":1,"nac_p":9,"nac_v":1,"sil":3,"sil_type":"perhour","gva":2,"sda":2,"mlat":[],"tisb":[],"messages":5327,"seen":0.2,"rssi":-22.4},
    {"hex":"495286","category":"A0","version":0,"sil_type":"unknown","mlat":[],"tisb":[],"messages":4341,"seen":259.3,"rssi":-27.4}
  ]
}
pi@SoftRF:~ $

It looks OK to me !!!
Good nite.

AC/.

lyusupov, 2020-01-05:
> I suppose that current 'flightaware' variant should be fine for SoftRF…

I've taken more close look into each one's README-json.md and can see, that 'flightaware'  is using similar but different 'dialect' of D0190 JSON rather than 'mutability' does.

I've updated [Example  6](https://github.com/lyusupov/SoftRF/wiki/Raspberry-Edition#example-6-ground-station-relay-ads-b-traffic-into-air-for-use-by-ogntp-compatible-airborne-receivers) with:
1. explicit 'stty' instruction ( not sure if you and others are aware of [this issue](https://github.com/lyusupov/SoftRF/issues/67#issuecomment-529942379) which may happen on different versions of Linux distributions for Raspberry Pi ) 
2. explicit reference onto 'mutability' version of **dump1090**

lyusupov, 2020-01-05:
## Below is a list of actions that I do and the results that I've got after these actions

1. I enable some debug print stuff [here](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/JSONHelper.cpp#L654),  [here](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/Platform_RPi.cpp#L613) and [there](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/Platform_RPi.cpp#L631)

2. I disable range check [here](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/Platform_RPi.cpp#L624):

```
      } else if (isValidFix() &&
                 Container[i].addr &&
                 Container[i].latitude  != 0.0 &&
                 Container[i].longitude != 0.0 &&
                 Container[i].altitude  != 0.0 &&
                 /* Container[i].distance < (ALARM_ZONE_NONE * 2) */ true ) {
```

3. make -f Makefile.RPi clean ; make -f Makefile.RPi

4. I do

```
pi@raspberrypi $stty -F /dev/ttyAMA0 9600 raw
pi@raspberrypi $ { echo "{class:SOFTRF,mode:RELAY,nmea:{private:true}}" ; cat /dev/ttyAMA0 ; } | sudo ./SoftRF

SoftRF-RPi FW.REV: 1.0-rc7 DEV.ID: D3374780
SX1276 RFIC is detected.
$GPGGA,091650.477,5600.5503,N,03822.2378,E,1,05,3.6,183.4,M,13.1,M,,0000*5B
$GPGSA,A,3,09,02,07,30,06,,,,,,,,5.2,3.6,3.7*3D
$GPGSV,3,1,12,09,87,225,28,23,63,104,,04,56,098,15,07,38,215,45*79
$GPGSV,3,2,12,16,36,079,,26,27,050,,06,25,252,41,02,25,296,36*7F
$GPGSV,3,3,12,03,18,155,15,29,09,358,,30,07,224,34,05,04,318,*75
$GPRMC,091650.477,A,5600.5503,N,03822.2378,E,000.0,258.2,210500,,,A*6F
< ... skipped ... >
```

5. I open one more terminal session and do this

```
pi@raspberrypi $ cat ~/SoftRF/aircraft.json | nc -N localhost 30007
```

and I see this output:

```
< ... skipped ... >
$GPGGA,092251.478,5600.5507,N,03822.2372,E,1,06,1.8,182.2,M,13.1,M,,0000*54
$GPGSA,A,3,03,09,02,07,30,06,,,,,,,2.6,1.8,1.9*3D
$GPGSV,3,1,12,09,87,166,27,23,61,105,,04,53,099,24,07,41,216,45*7A
$GPGSV,3,2,12,16,37,076,,26,26,047,,02,25,293,35,06,23,250,41*74
$GPGSV,3,3,12,03,15,155,13,30,09,224,29,29,08,356,,05,07,318,*70
$GPRMC,092251.478,A,5600.5507,N,03822.2372,E,000.0,258.2,210500,,,A*68
$GPVTG,258.2,T,,M,000.0,N,000.0,K,A*00
155BD8 55.609619 37.388439 1021.080017 1 1024 9
42437B 55.997860 37.597458 1455.420044 1 1536 9
424567 56.319759 37.113560 1905.000000 1 -832 9
424921 55.751900 36.862122 2125.979980 1 -960 9
4248E8 56.078793 36.015259 7322.819824 1 2304 9
424284 55.450317 37.678291 1844.040039 1 1792 9
42428B 55.489330 37.792622 1036.319946 1 1920 9
06A139 55.299679 37.758171 10660.379883 1 0 9
$PSRFO,958900971,D85B15032C4A8E8139E36AEA02158F4253829007
155BD8 55.609619 37.388439 1021.080017 1 1024 9
$GPGGA,092252.478,5600.5507,N,03822.2372,E,1,06,2.0,182.2,M,13.1,M,,0000*5C
$GPGSA,A,3,23,09,02,07,30,06,,,,,,,3.4,2.0,2.8*35
$GPGSV,3,1,12,09,87,161,27,23,61,105,14,04,53,099,24,07,41,216,45*78
$GPGSV,3,2,12,16,37,076,,26,26,047,,02,25,293,35,06,23,250,40*75
$GPGSV,3,3,12,03,15,155,12,30,09,224,28,29,08,356,,05,07,318,*70
$GPRMC,092252.478,A,5600.5507,N,03822.2372,E,000.0,258.2,210500,,,A*6B
$GPVTG,258.2,T,,M,000.0,N,000.0,K,A*00
$GPGGA,092253.478,5600.5507,N,03822.2371,E,1,06,2.0,182.2,M,13.1,M,,0000*5E
$PSRFO,958900972,7B43420B037C6159105F3A7C5B54C35804781C6C
[... 139 lines cut ...]
42428B 55.489330 37.792622 1036.319946 1 1920 9
```

7. then I open glidertracker.org and see of how it looks like there:

[image]

8. Then I do this:

```
pi@raspberrypi $ echo quit | nc -N localhost 30007
```

to properly terminate 'SoftRF' UNIX process:

```
< ... skipped ... >
$GPRMC,094139.481,A,5600.5489,N,03822.2280,E,000.0,258.2,210500,,,A*6E
$GPVTG,258.2,T,,M,000.0,N,000.0,K,A*00
$GPGGA,094140.481,5600.5489,N,03822.2280,E,1,07,1.6,187.6,M,13.1,M,,0000*52
$GPGSA,A,3,23,09,02,07,30,05,06,,,,,,2.9,1.6,2.5*34
$GPGSV,3,1,12,09,79,120,21,23,52,108,18,07,50,218,47,04,45,103,20*76
Program termination.
pi@raspberrypi $
```

lyusupov, 2020-01-05:
This is ZIP of aircraft.json file that I used in the example above:

[aircraft.zip](https://github.com/lyusupov/SoftRF/files/4023248/aircraft.zip)

Hope this helps you...

acasadoalonso, 2020-01-05:
Linar,
That is very helpful indeed, I will try with the FligthAware version of the dump1090 and let you know, so you can share it with some other people.
I saw that the FA version does not support by default the http to the port 8080 and relays on a file like /tmp/aircraft.json to be passed to the web servers or to SoftRF thru the “nc” command.
Thanks again for your great support.

AC/.

lyusupov, 2020-01-05:
Read that I've said [above](https://github.com/lyusupov/SoftRF/issues/74#issuecomment-570891513).

**Details**: FA uses 2 altitudes (geo and pressure) and 3 speeds (tas, ias, gs). MUTABILITY uses 1 altitude and 1 speed. JSON tokens for them are different. 
**FA JSON will be just ignored** by SoftRF because of [this condition](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/JSONHelper.cpp#L580)

acasadoalonso, 2020-01-05:
I see thanks, I misunderstood the different dialect comment. 
Thanks again.

AC/.

acasadoalonso, 2020-01-05:
Great job Linar ... your support is second to none !!!!

I will submit some enhancement to the documentation like now the stretch Raspbian distro is using automaker 1.16 and the libraries are built using automaker 1.5, so autoreconf needs to be run, essentially commenting my experiences 

Again thanks so much.

AC/.

acasadoalonso, 2020-01-05:
[image]

Great support indeed.
AC/.

closed 2020-01-05 by acasadoalonso