lyusupov/SoftRF #76: SoftRF show its own deviceID as traffic

8lazo8, 2020-01-09:
## Hardware

- [ ]   Standalone
- [x]   Prime Mark II
- [ ]   Dongle
- [ ]   SkyView EZ
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   UAT978 module
- [ ]   Flight Recorder

[image]

## Firmware version

[image]

## Firmware settings

[image]

## Describe the bug (or ask a question)

I am a new SoftRF user, today I went out for a test on the ground.
Next week I plan to make an in flight test.

When the Prime MKII is near to an aircraft - 2-300 meters or closer - equiped with OGN tracker periodically it seems to "receive" its own tracker ID as foreign tracker and it is listed as traffic in the Skyview and in xcsoar as well.  The phenomenon is intermittent.
It tested this with two different aircrafts.

After the aircrafts departured the "self reflection" disappeared only the real aircraft was seen on the screen.

## To Reproduce

Te be close to an OGN tracker equipped aircraft, maximum 2-300 meters.

## Expected behavior

To receive/display only the foreign aircraft deviceID but not itself.

Ben64Smith, 2020-01-09:
From memory I think paweł added rebroadcast functionality in the later versions of the OGN tracker, this is probably the cause of your issue.

8lazo8, 2020-01-09:
> 
> 
> From memory I think paweł added rebroadcast functionality in the later versions of the OGN tracker, this is probably the cause of your issue.

This means that the foreign aircraft rebroadcasted my deviceID?  I am pretty sure it was not SoftRF device inside.

Does this rebroadcast function has anything to do to proximity of the trackers? So its not active on larger distances? I mean I should have to receive it all the time. And it also would cause alerts.
Or in this case the self deviceID should be filtered?

lyusupov, 2020-01-09:
@Ben64Smith 
Correct. I also expect that this likely happens due to re-broadcast feature of OGN tracker.
I don't remember if I have seen this with my build of STM32 OGN tracker on a breadboard.
It probably depends on particular revision of OGN firmware, settings and certain conditions to be met.

@8lazo8 
Could you connect your Prime Mk2 to a PC and **capture a NMEA log on 38400,8*N1** ? 
If the is a 'ghosting traffic' - we will likely see it in a PFLAA sentence.

8lazo8, 2020-01-09:
> 
> 
> @Ben64Smith
> Correct. I also expect that this likely happens due to re-broadcast feature of OGN tracker.
> I don't remember if I had seen this with my build of STM32 OGN tracker on a breadboard.
> It probably depends on particular revision of OGN firmware, settings and certain conditions has to be met.
> 
> @8lazo8
> Could you connect your Prime Mk2 to a PC and **capture a NMEA log on 38400,8*N1** ?
> If there is a 'ghosting traffic' - we will likely see it in a PFLAA sentence.

Yes I can do that. You need the capture when "ghosting" can be seen on the screen?

I have two Mk2-s but between them this never happened. To reproduce the phenomenon I will need to go for another field trip.
I will capture the log ASAP whichever is needed.

lyusupov, 2020-01-09:
No rush. Take it when able. ZIP , then drag and drop here.

> You need the capture when "ghosting" can be seen on the screen?

Yes.

8lazo8, 2020-01-11:
> 
> 
> No rush. Take it when able. ZIP , then drag and drop here.
> 
> > You need the capture when "ghosting" can be seen on the screen?
> 
> Yes.

Hi,
Today I managed to collect logs.  After all I could not take it from the serial but I hope xcsoar NMEA dump will be good as well. 
There were lots of "ghostings".
[2020-01-11_10-53.zip](https://github.com/lyusupov/SoftRF/files/4049437/2020-01-11_10-53.zip)

lyusupov, 2020-01-11:
Thanks!

I've committed https://github.com/lyusupov/SoftRF/commit/b0ddbac7e7af06a9929581df677d2e3aefa4477a and https://github.com/lyusupov/SoftRF/commit/6e4e9bb6ee95fe397c8b22baf141dc555ef2e9b2

[Build](https://github.com/lyusupov/SoftRF/tree/master/software/firmware/source#esp32) and update your **Prime MkII**(s).

Those 2 commits above should fix this issue. Please, test and confirm.

8lazo8, 2020-01-11:
> 
> 
> Thanks!
> 
> I've committed [b0ddbac](https://github.com/lyusupov/SoftRF/commit/b0ddbac7e7af06a9929581df677d2e3aefa4477a) and [6e4e9bb](https://github.com/lyusupov/SoftRF/commit/6e4e9bb6ee95fe397c8b22baf141dc555ef2e9b2)
> 
> [Build](https://github.com/lyusupov/SoftRF/tree/master/software/firmware/source#esp32) and update your **Prime MkII**(s).
> 
> Those 2 commits above should fix this issue. Please, test and confirm.

Update has been done, will test it soon.

8lazo8, 2020-01-17:
There is IMC in the area since a week, hopefully next week it will good enough for a test.

lyusupov, 2020-01-27:
**Resolution:** fixed in https://github.com/lyusupov/SoftRF/commit/b0ddbac7e7af06a9929581df677d2e3aefa4477a and https://github.com/lyusupov/SoftRF/commit/6e4e9bb6ee95fe397c8b22baf141dc555ef2e9b2

closed 2020-01-27 by lyusupov