lyusupov/SoftRF #77: t-beam v1.0 does not shut down correctly

zodiac-maniac, 2020-01-14:
**ATTENTION** ! 

- Use this form to report bugs in released SoftRF firmware, hardware and web pages only ;
- Issue reports from anonymous users will be most likely invalidated. Reasons for that are well explained [here](https://berthub.eu/articles/posts/anonymous-help/) ;
- Prior to asking a question, look into ["Closed issues"](https://github.com/lyusupov/SoftRF/issues?q=is%3Aissue+is%3Aclosed) area to see if it was answered before ;
- No any "ideas" or new feature requests, please ! New feature requests do not subject for taking into consideration unless they come from co-developers ;
- "Enhancements" of existing features are allowed only when you are ready to implement them by yourself ;

Recommended:
- screenshot or photo width - from 600 to 800 pixels ;
- aspect ratio - 4:3, 3:4, 16:9 or 9:16.
- substitute 'space' by 'x' character to mark a hardware checkbox

--------------------

## Hardware

- [ ]   Standalone
- [x ]   Prime Mark II
- [ ]   Dongle
- [ ]   SkyView EZ
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   UAT978 module
- [ ]   Flight Recorder

< photo of your hardware > (**required**)

## Firmware version
RC7
<  screenshot of WebUI status page > (**required**)

## Firmware settings
Legacy protocol, UDP for flymap
<  screenshot of WebUI settings page > (**required**)

## Describe the bug (or ask a question)
Starting the t-beam often is difficult. It don't respond to power on. You have to click a lot to start it. When removing the battery, power up works for a few times and then same problem. 
When power down the display shows shut down and device appears to be down but checking WiFi it's still active and WiFi is working. Only WiFi, no GPS or tx/Rx. When checking webpage it says 0.35V or so but the Webinterface works together with wifi.
Strange behaviour and when power it up and then down it seems io be down and WiFi is down as well.

## To Reproduce

## Expected behavior
Should work like my v0.7 t-beam but it doesn't. But when flarm is working, it worked. But power up and power down is weird.

lyusupov, 2020-01-14:
Units that I have - they work fine to me.

**Resolution**: unable to reproduce.

closed 2020-01-14 by lyusupov

zodiac-maniac, 2020-01-15:
Yeah, that easy it is...and others and I still have that problem. Maybe you should try to not remove the battery for longer time.