lyusupov/SoftRF #84: P3I problem

kaelsaes, 2020-06-07:
**ATTENTION** ! 

- Use this form to report bugs in released SoftRF firmware, hardware and web pages only ;
- Issue reports from anonymous users will be most likely invalidated. Reasons for that are well explained [here](https://berthub.eu/articles/posts/anonymous-help/) ;
- Prior to asking a question, look into ["Closed issues"](https://github.com/lyusupov/SoftRF/issues?q=is%3Aissue+is%3Aclosed) area to see if it was answered before ;
- No any "ideas" or new feature requests, please ! New feature requests do not subject for taking into consideration unless they come from co-developers ;
- "Enhancements" of existing features are allowed only when you are ready to implement them by yourself ;

Recommended:
- screenshot or photo width - from 600 to 800 pixels ;
- aspect ratio - 4:3, 3:4, 16:9 or 9:16.
- substitute 'space' by 'x' character to mark a hardware checkbox

--------------------

## Hardware

- [ ]   Standalone
- [ x]   Prime Mark II
- [ ]   Dongle
- [ ]   SkyView EZ
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   UAT978 module
- [ ]   Flight Recorder

< photo of your hardware > (**required**)

## Firmware version

<  screenshot of WebUI status page > (**required**)

## Firmware settings

<  screenshot of WebUI settings page > (**required**)

## Describe the bug (or ask a question)

PAW Rosetta and SoftRF P3I don't see each other. 
Stratux Europe also can't see SoftRF with P3I protocol. 

## To Reproduce

## Expected behavior

wadoadi, 2020-06-07:
I can see my Prime II in pilotaware but not in Stratux EU.

kaelsaes, 2020-06-07:
Thanks.

After selecting UK(869.52 MHz) my PAW Rosetta can see my Prime Mark II.
But not vice versa.

Stratux and MK II still can't see each other.

CT2IUV, 2020-06-07:
Adjust the PPM on stratux, and will work

Regards

lyusupov, 2020-06-07:
- anonymous user ;
- imcomplete issue report.

**Resolution**: invalid

closed 2020-06-07 by lyusupov

wadoadi, 2020-06-08:
> Adjust the PPM on stratux, and will work Regards
> […](#)

What PPM value should we try?