lyusupov/SoftRF #85: Skytraxx Flarm is not recognized

ThomasMPunkt, 2020-06-16:
--------------------

## Hardware

- [ ]   Standalone
- [X]   Prime Mark II
- [ ]   Dongle
- [ ]   SkyView EZ
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   UAT978 module
- [ ]   Flight Recorder

< photo of your hardware > (**required**)

## Firmware version

[image]

## Firmware settings

[image]

## Describe the bug (or ask a question)

I installed v1.0-rc7 on a TTGO. Unfortunately I do not receive any signals from Skytraxx Varios. The Skytraxx Varios are received by an OGN station nearby. 

Protocol is set to "legacy".
If I change the protocol to "FANET", the Skytraxx Vario is displayed.
Skytraxx also sends FLARM signals in addition to FANET. How can I display this?

## To Reproduce

## Expected behavior

GithubReinart, 2020-06-16:
Hello

I guess this is because of Skytraxx devices and SoftRF using different frequencies for FLARM (868.2 MHz/868.4 MHz)

Regards
Manfred

lyusupov, 2020-06-16:
photo of your hardware ?

lyusupov, 2020-06-16:
* incomplete issue report.

**Resolution**: invalid

lyusupov, 2020-06-16:
the Skytraxx is likely required to be in motion in order to transmit FLARM packets, read [this Gitter thread](https://gitter.im/lyusupov/SoftRF?at=5db7e829a03ae1584f2b138f)

mreinart, 2020-06-16:
The Skytraxx will send FLARM signals once it has acquired GPS fix and is in flight (recording) mode. This is either by moving around or by manually triggering the recording. 
Double checked this with my own Skytraxx device and an OGN base station operated on a RaspBerry PI which receives signals from the Skytraxx under these conditions.
Also observed that the OGN-BaseStation 'sees' the SkyTraxx via FLARM (not FANET) but SkyTraxx is not seen by a SofTRF unit operated in Legacy mode.

Difference in frequency used by Skytraxx and OGN can also be checked by changing the frequency in SoftRF e.g. in RPi installation running in parallel.

lyusupov, 2020-06-16:
If Skytraxx does not transmit FLARM exactly the same way as genuine FLARM does - it is the issue of Skytraxx, not **SoftRF** .
There is no any statements in SoftRF documentation that it receives Skytraxx FLARM transmit sessions. Thus, there is no bug here.

If someone wants to make an improvement - I can assign this ticket onto the volunteer and will expect to receive a PR sometime. If not - I will close this ticket soon.

**Resolution**: unable to reproduce

ThomasMPunkt, 2020-06-16:
> 
> 
> the Skytraxx is likely required to be in motion in order to transmit FLARM packets, read [this Gitter thread](https://gitter.im/lyusupov/SoftRF?at=5db7e829a03ae1584f2b138f)

Skytraxx devices that fly are not shown.

mreinart, 2020-06-16:
> If Skytraxx does not transmit FLARM exactly the same way oas genuine FLARM does - it is the issue of Skytraxx, **not SoftRF** .
> There are no any statements in SoftRF documentation that it receives Skytraxx FLARM transmit sessions. Thus, there is no bug here.

Dear Linar, why are you so sensible and also so fast in closing issue reports on formal aspects like missing photo (while it was clear that the HW is a new self contained TTGO device)? 

None blames SoftRF having a bug or claiming "Skytraxx-Compatibility".
It is just a recognized matter of fact that there is a difference in transmission and reception behaviour between Skytraxx devices (transmitting only FLARM for licensing reasons- but being received by OGN base stations) and the way SoftRF operates in Legacy mode. 
Either only seems to use one fixed frequency out of the set {868.2,  868.4}. There only seem to be different - non-overlapping operating decisions. In my environment with SoftRF, SkyTraxx, OGN base station on RPi and my own flavor of FANET-Groundstation on RPi I can reproduce this. 
In a separate small project on RPi I was able to receive and decode FLARM-Signals form SoftRF and Legacy-Track-Packets from SoftRF devices by toggleing the receiver frequency between the two values listed above. 
This could also be a solution approach for SoftRF. Would be willing to send a change proposal for SoftRF after clarifying the questions why SoftRF currently uses only one frequency seemingly in EU band.

lyusupov, 2020-06-16:
@mreinart 
> ... Would be willing to send a change proposal for SoftRF ...

Nice... But, is there any progress with [this](https://github.com/lyusupov/SoftRF/issues/67#issuecomment-530263695) first ?

GithubReinart, 2020-06-16:
@lyusupov I guess this falls under 'improvement of "legacy" protocol compatibility'

lyusupov, 2020-06-16:
@mreinart
Ok. And what about 2.3 you wrote that "will invest time in these areas" in 2019 ? Have you done it already ?

GithubReinart, 2020-06-16:
@lyusupov on 2.3 I investigated options as I also am building a FANET base station with weather data and messaging. 
In this area there is much variety WRT remote manageability - which does not seem very crucial from SoftRF-only perspective. As you mentioned yourself there does not seem to be a big clientele for RPi-based SoftRF installations- i.e. as SoftRF is focussed on a GPS fix - which is not so much of a requirement for base stations where the own geoloc could be configured. Thus this is not finished as I am working on the FANET base station deployment with remote management capabilities.

lyusupov, 2020-06-18:
Ok, so far I see no volunteers from numerous subscribers of this GitHub project who could and would like to devote his/her time just to make the issue submitter happy.

It was already indicated above that this report is not a bug but instead is a 'request for feature/improvement' which are not allowed here unless someone wants to implement that.

**Resolution**: invalid

closed 2020-06-18 by lyusupov