lyusupov/SoftRF #86: Error on fanet position while sending to lk8000.

danielvelezbravo, 2020-07-07:
**ATTENTION** ! 

- Use this form to report bugs in released SoftRF firmware, hardware and web pages only ;
- Issue reports from anonymous users will be most likely invalidated. Reasons for that are well explained [here](https://berthub.eu/articles/posts/anonymous-help/) ;
- Prior to asking a question, look into ["Closed issues"](https://github.com/lyusupov/SoftRF/issues?q=is%3Aissue+is%3Aclosed) area to see if it was answered before ;
- No any "ideas" or new feature requests, please ! New feature requests do not subject for taking into consideration unless they come from co-developers ;
- "Enhancements" of existing features are allowed only when you are ready to implement them by yourself ;

Recommended:
- screenshot or photo width - from 600 to 800 pixels ;
- aspect ratio - 4:3, 3:4, 16:9 or 9:16.
- substitute 'space' by 'x' character to mark a hardware checkbox

--------------------
[image]
[image]
[image]

## Hardware

- [ ]   Standalone
- [ ]   Prime Mark II
- [ ]   Dongle
- [ ]   SkyView EZ
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   UAT978 module
- [ ]   Flight Recorder

< photo of your hardware > (**required**)

SoftRF-firmware-v1.0-rc7-ESP32

<  screenshot of WebUI status page > (**required**)

## Firmware settings

<  screenshot of WebUI settings page > (**required**)

## Describe the bug (or ask a question)

Position of nearby devices is 300 mts east of real position only when viewed on lk8000, but seems ok when broadcasted to internet, or received by other units.

## To Reproduce

Probably you need to be near my place. 6.1791 -75.5818 

## Expected behavior
Position of nearby devices should be nearby (as 0km away).

lyusupov, 2020-07-07:
1) your SoftRF Model ?  
   Use: "  substitute 'space' by 'x' character to mark a hardware checkbox  "  recommendation specified in the template.

2) < photo of your hardware > (required)

3) < screenshot of WebUI status page > (required)

Incomplete issue report.
**Resolution**: invalid

4) **Describe the bug**: "Position of nearby devices is 300 mts east of real position only when viewed on lk8000"

Why do you file the ticket here but not in [LK8000 GitHub](https://github.com/LK8000/LK8000) or [Postfrontal support forum](https://postfrontal.com/forum/default.asp?CAT_ID=11) ?

**Resolution**: invalid

5) **To Reproduce**:  " Probably you need to be near my place. 6.1791 -75.5818** "

**Resolution**: unable to reproduce

closed 2020-07-07 by lyusupov

brunotl, 2020-07-07:
add NMEA log to see if problem come from SoftRF or from LK8000.

danielvelezbravo, 2020-07-07:
> Sorry.... Really newby so lots of mistakes.  I found your SoftRF github
> web and no idea how to contact you. Created a Github account and tried to
> learn Github only to report...
>
> So far:
>
> 1. It´s definitely not a Lk800 error. As its not present on Airwhere, and
> its already fixed on GxAir.  So I already reported the error to Gerald and
> he shorted it out with a new firmware update for his beta software.  Then I
> was messing around to try to see how Airwhere, GxAir and SoftRF talked to
> each other, and "boom"... your software has the bug.
>
> So I was trying to make a RealTerm "Serial" port monitoring to record the
> sentences that are processed, but got some garbage... Maybe I need to "set"
> the softrf to serial output?
>
> I can grab some screenshots, or we can "connect" and I can show you what
> is happening.
>
> Gerald connected throught TeamViewer to my pc and he grabbed some
> sentences from the serial port and fixed it on his code.
>
> You could do the same.  I´m willing to help but I´m not a programmer so
> I´m not playing by the rules of a programmer or GitHub yet.
>
> The Hardware is a TTGO T-beam Ver 1.1. Connected BLE to Android.
>
> My whatsapp +573104487870 or Facebook Messenger danielvelezbravo.
> We can chat through there.

[image: image.png]
[image: image.png]

$GPRMC,221932.00,A,0610.74697,N,07534.91367,W,4.654,,070720,,,A*69
$GPGGA,221932.00,0610.74697,N,07534.91367,W,1,06,1.37,1599.4,M,2.7,M,,*42
$GPGSA,A,3,15,13,29,02,12,05,,,,,,,2.30,1.37,1.85*03
$PFLAA,3,5,325,2,3,04C002!FAN_04C002,0,,0,0.0,6*6A
$PFLAA,3,0,330,-1,3,076E98!FAN_076E98,0,,0,0.1,6*44
$PFLAU,2,1,2,1,3,89,2,2,325,04C002*3D
~ ÅÄÙ9  ©Ë~~e ˇˇˇˇˇˇˇˇSoftRF  SoftRF             ˇô~~
 ›‘∏dﬂ @Ωôà @ FADDD4B8 è~~ 
'ø~~ ¿dﬂÄ  ©à    FA04C002 õ~~ nòd‡Ä  ôà    FA076E98 \~*8E04C002581FA01E933A16AF06B9;
*8E04C002581FA40CFFA59499F704;
*8E04C0020C181C340F0C32EFD68A;
*8E04C00299000100300400509786;
*8E076E98581FA01E933A173331A3;
*8E076E98581FA40D01A5946B560C;
*8E076E980C181C37D85E78EF2341;
*8E076E9899000100300400335495;
$GPRMC,221933.00,A,0610.74676,N,07534.91359,W,0.166,,070720,,,A*68
$GPGGA,221933.00,0610.74676,N,07534.91359,W,1,06,1.37,1599.6,M,2.7,M,,*43
$GPGSA,A,3,15,13,29,02,12,05,,,,,,,2.30,1.37,1.85*03
$PFLAA,3,5,325,2,3,04C002!FAN_04C002,0,,0,0.0,6*6A
$PFLAA,3,5,328,-1,3,076E98!FAN_076E98,0,,0,0.1,6*48
$PFLAU,2,1,2,1,3,89,2,2,325,04C002*3D
~ ÅÄı9  ò€~~e ˇˇˇˇˇˇˇˇSoftRF  SoftRF             ˇô~~
 ›‘∏dﬂ @Ωôà   FADDD4B8 ÕÀ~~ 
'ø~~ ¿dﬂÄ  ©à    FA04C002 õ~~ nòd‡Ä  ôà    FA076E98 \~*8E04C002581FA01E933A16AF06B9;
*8E04C002581FA40CFFA59499F704;
*8E04C0020C181C340F0C32EFD68A;
*8E04C00299000100300400509786;
*8E076E98581FA01E933A173331A3;
*8E076E98581FA40D01A5946B560C;
*8E076E980C181C37D85E78EF2341;
*8E076E9899000100300400335495;
$GPRMC,221934.00,A,0610.74696,N,07534.91384,W,2.210,,070720,,,A*61
$GPGGA,221934.00,0610.74696,N,07534.91384,W,1,06,1.37,1600.2,M,2.7,M,,*4D
$GPGSA,A,3,15,13,29,02,12,05,,,,,,,2.30,1.37,1.85*03
$PFLAA,3,5,325,1,3,04C002!FAN_04C002,0,,0,0.0,6*69
$PFLAA,3,5,328,-2,3,076E98!FAN_076E98,0,,0,0.1,6*4B
$PFLAU,2,1,2,1,3,89,2,1,325,04C002*3E
~ ÅÄˆ9  Àé~~e ˇˇˇˇˇˇˇˇSoftRF  SoftRF             ˇô~~
 ›‘∏dﬂ @Ωôà   FADDD4B8 ∑a~~ 
'ø~~ ¿dﬂÄ  ©à    FA04C002 õ~~ nòd‡Ä  ôà    FA076E98 \~*8E04C002581FA01E933A16AF06B9;
*8E04C002581FA40CFFA59499F704;
*8E04C0020C181C340F0C32EFD68A;
*8E04C00299000100300400509786;
*8E076E98581FA01E933A173331A3;
*8E076E98581FA40D01A5946B560C;
*8E076E980C181C37D85E78EF2341;
*8E076E9899000100300400335495;
$GPRMC,221935.00,A,0610.74669,N,07534.91358,W,2.322,,070720,,,A*61
$GPGGA,221935.00,0610.74669,N,07534.91358,W,1,06,1.37,1602.2,M,2.7,M,,*4F
$GPGSA,A,3,15,13,29,02,12,05,,,,,,,2.30,1.37,1.85*03
$PFLAA,3,5,325,0,3,04C002!FAN_04C002,0,,0,0.0,6*68
$PFLAA,3,5,329,-4,3,076E98!FAN_076E98,0,,0,-0.1,6*61
$PFLAU,2,1,2,1,3,89,2,0,325,04C002*3F
~ ÅÄ˜9  ˙Ω~~e ˇˇˇˇˇˇˇˇSoftRF  SoftRF             ˇô~~
 ›‘∏dﬂ @Ω©à   FADDD4B8 è:~~ 
eü~~ ¿dﬂÄ  ©à    FA04C002 õ~~ nòd·Ä  ôà    FA076E98 Cƒ~*8E04C002581FA01E933A16AF06B9;
*8E04C002581FA40CFFA59499F704;
*8E04C0020C181C340F0C32EFD68A;
*8E04C00299000100300400509786;
*8E076E98581FA01E953A171FC825;
*8E076E98581FA40D01A5946B560C;
*8E076E980C181C37D85E78EF2341;
*8E076E98990001005804009272E6;
$GPRMC,221936.00,A,0610.74628,N,07534.91300,W,2.389,,070720,,,A*6B
$GPGGA,221936.00,0610.74628,N,07534.91300,W,1,05,2.05,1603.7,M,2.7,M,,*41
$GPGSA,A,3,15,13,29,02,12,,,,,,,,4.54,2.05,4.05*0D
[... 410 lines cut ...]
*8E04C00299000100300400509786;
*8E076E98581FA01E933A173331A3;
*8E076E98581FA40CFFA594FA3417;
*8E076E980C181C37D85E78EF2341;
*8E076E9899000100300400335495;
$GPRMC,222107.00,A,0610.74805,N,07534.91577,W,1.078,,070720,,,A*6B
$GPGGA,222107.00,0610.74805,N,07534.91577,W,1,06,1.38,1594.2,M,2.7,M,,*49
$GPGSA,A,3,15,13,29,02,12,05,,,,,,,2.31,1.38,1.86*0E
$PFLAA,3,5,328,5,3,04C002!FAN_04C002,0,,0,0.0,6*60
$PFLAA,3,5,332,5,3,076E98!FAN_076E98,0,,0,0.0,6*6B
$PFLAU,2,1,2,1,3,89,2,5,329,04C002*36
~ ÅÄG:  îì~~e ˇˇˇˇˇˇˇˇSoftRF  SoftRF             ˇô~~
 ›‘∏d‡ @ªôà   FADDD4B8 &~~ 
»N~~ ¿d‡Ä  ôà    FA04C002 e†~~ nòd‡Ä  ôà    FA076E98 \~*8E04C002581FA01E933A16AF06B9;
*8E04C002581FA40D01A59408951F;
*8E04C0020C181C340F0C32EFD68A;
*8E04C00299000100300400509786;
*8E076E98581FA01E933A173331A3;
*8E076E98581FA40CFFA594FA3417;
*8E076E980C181C37D85E78EF2341;
*8E076E9899000100300400335495;
$GPRMC,222108.00,A,0610.74801,N,07534.91584,W,0.929,,070720,,,A*60
$GPGGA,222108.00,0610.74801,N,07534.91584,W,1,06,1.38,1594.7,M,2.7,M,,*4B
$GPGSA,A,3,15,13,29,02,12,05,,,,,,,2.31,1.38,1.86*0E
$GPRMC,222109.00,A,0610.74809,N,07534.91603,W,1.397,,070720,,,A*6B

lyusupov, 2020-08-02:
Read this: https://github.com/lyusupov/SoftRF/commit/ec74364c7bcf01c2a377d7943dba23d5783e2bf1 and   that: https://github.com/lyusupov/SoftRF/commit/af38d14ff1534c2b6b99d0c23e74b5663276d4da