lyusupov/SoftRF #87: Prime Mark II,  1.0-rc8 , GNSS not detected when OLED attached

bluefix, 2020-07-09:
--------------------

## Hardware

- [ ]   Standalone
- [x]   Prime Mark II
- [ ]   Dongle
- [ ]   SkyView EZ
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   UAT978 module
- [ ]   Flight Recorder

[image]

## Firmware version
SoftRF-ESP32 FW.REV: 1.0-rc8

[image]

[image]

## PROBLEM 
**when OLED (21, 22) is attached, GNSS is not detected during boot**
 
**no change if BMP280 (pins 2, 13) is attached, or not** 
 
**without OLED boot is successful and device is sending proper legacy packages.**
 

[image]

## boot log with OLED and BMP280

SoftRF-ESP32 FW.REV: 1.0-rc8 DEV.ID: b582f404
Copyright (C) 2015-2020 Linar Yusupov. All rights reserved.

Reset reason: 0
POWERON_RESET
Free heap size: 210784
Vbat power on reset

EEPROM version: 94
SX1276 RFIC is detected.
BMP280 barometric pressure sensor is detected.
Temperature = 25.58 *C
Pressure = 100387.64 Pa
Approx altitude = 78.34 m

INFO: TTGO T-Beam rev. 016 is detected.
Flash memory ID: 684016
Hostname: SoftRF-82f404
Wait for WiFi connection.
....................
Can not connect to WiFi station. Go into AP mode.
Setting soft-AP configuration ... Ready
Setting soft-AP ... Ready
IP address: 192.168.1.1
UDP server has started at port: 12389
HTTP server has started at port: 80
$PFLAU,0,0,0,1,0,,0,,,*4F
$PGRMZ,256,f,3*1A
$PFLAU,0,0,0,1,0,,0,,,*4F
$PGRMZ,255,f,3*19
$PFLAU,0,0,0,1,0,,0,,,*4F
$PGRMZ,255,f,3*19
$PFLAU,0,0,0,1,0,,0,,,*4F
$PGRMZ,256,f,3*1A
$PFLAU,0,0,0,1,0,,0,,,*4F
$PGRMZ,255,f,3*19
$PFLAU,0,0,0,1,0,,0,,,*4F
$PGRMZ,255,f,3*19
$PFLAU,0,0,0,1,0,,0,,,*4F
$PGRMZ,256,f,3*1A
$PGRMZ,254,f,3*18
$PFLAU,0,0,0,1,0,,0,,,*4

## boot log with BMP280 only

SoftRF-ESP32 FW.REV: 1.0-rc8 DEV.ID: b582f404
Copyright (C) 2015-2020 Linar Yusupov. All rights reserved.

Reset reason: 0
POWERON_RESET
Free heap size: 210784
Vbat power on reset

EEPROM version: 94
SX1276 RFIC is detected.
BMP280 barometric pressure sensor is detected.
Temperature = 25.97 *C
Pressure = 100382.73 Pa
Approx altitude = 78.75 m

INFO: TTGO T-Beam rev. 08 is detected.
Flash memory ID: 684016
INFO: GNSS module HW version: 00040007
INFO: GNSS module FW version: 7.03 (45969)
Hostname: SoftRF-82f404
Wait for WiFi connection.
....................
Can not connect to WiFi station. Go into AP mode.
Setting soft-AP configuration ... Ready
Setting soft-AP ... Ready
IP address: 192.168.1.1
UDP server has started at port: 12389
HTTP server has started at port: 80
$GPRMC,142858.00,A,5230.38022,N,01314.25021,E,2.366,,090720,,,A*73
$GPGGA,142858.00,5230.38022,N,01314.25021,E,1,06,2.06,72.2,M,42.4,M,,*65
$GPGSA,A,3,11,08,32,10,22,27,,,,,,,3.86,2.06,3.27*00
$PFLAU,0,1,2,1,0,,0,,,*4C
$PGRMZ,257,f,3*1B
$GPRMC,142910.00,A,5230.37194,N,01314.25861,E,1.083,,090720,,,A*7A
$GPGGA,142910.00,5230.37194,N,01314.25861,E,1,06,1.66,72.5,M,42.4,M,,*65
$GPGSA,A,3,11,08,32,10,22,27,,,,,,,3.35,1.66,2.90*00
$GPRMC,142911.00,A,5230.37097,N,01314.25885,E,1.460,,090720,,,A*7A
$PFLAU,0,1,2,1,0,,0,,,*4C
$PGRMZ,257,f,3*1B

lyusupov, 2020-07-09:
> INFO: TTGO T-Beam rev. 016 is detected.

There is **NO** such thing as T-Beam rev 16, but there **IS** TTGO LoRa that has rev 16.

Your board is misdetected so incorrect GNSS Rx/Tx pins have been selected.

The reason for that is likely that you've eneterd SLEEP mode proir to this boot.
NEVER user SLEEP when OLED is connected - **'Noname' OLED has known hardware issue that it blocks I2C bus access when switched into 'power save' mode**. 

T-Beam V10 does not function properly when access to AXP192 over I2C is blocked.
And we see this happened in your log.

**Solution**: NEVER use SLEEP mode with your Prime Mk2 with OLED. Use full POWER OFF instead.
Be patient to keep holding power button AT LEAST 7 SECONDS to execute 'power off' sequence.

lyusupov, 2020-07-09:
**Known issues** section of [**Release notes**](https://github.com/lyusupov/SoftRF/releases/tag/1.0-rc8) has been updated.

**Resolution**: deferred

lyusupov, 2020-07-09:
And one more notice for you to concern:

Neither BMP280 nor OLED are offically supported components of [**Prime Mk II**](https://github.com/lyusupov/SoftRF/wiki/Prime-Edition-MkII).
 I<sup>2</sup>C bus feature is a 'privilege' of [**Standalone Edition**](https://github.com/lyusupov/SoftRF/wiki/Standalone-Edition) & [**LoRa**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-LoRa-module) builders.

So filing of an  I<sup>2</sup>C device issue onto a **Prime Mk II** is not allowed here.

commit af38d14f (2020-08-02): ESP32 1.0-RC8 SoftRF firmware binary is updated due to critical fix #87 [skip ci]
closed 2020-08-02 by lyusupov