lyusupov/SoftRF #89: SoftRF show its own deviceID as traffic - this time Legacy protocol

BeenThereScrewedUp, 2020-08-04:
## Hardware

- [ ]   Standalone
- [x]   Prime Mark II
- [ ]   Dongle
- [ ]   SkyView EZ
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   UAT978 module
- [ ]   Flight Recorder

[image]

## Firmware version

Firmware SoftRF-firmware-v1.0-rc8-ESP32 after critical fix is used.

[image]

## Firmware settings

[image]

## Describe the bug (or ask a question)

During flight and on the ground, the unit picks up its own device ID as traffic. This happened without the unsupported TTGO OLED display as well.
I have no idea how the device can pick up its own radio transmission - i got this behaviour at 5000' AGL as well as during approaches and on the ground. If it picks up its own ID this continues for a minute or two and the stops for some time. Inside my office this does not occur. I even got a warning with the device sitting on my porch (lat/long substituted by 1234) - see serial log below:

SoftRF-ESP32 FW.REV: 1.0-rc8 DEV.ID: b224821c
Copyright (C) 2015-2020 Linar Yusupov. All rights reserved.

Reset reason: 0
POWERON_RESET
Free heap size: 210596
Vbat power on reset

EEPROM version: 94
SX1276 RFIC is detected.
WARNING! Barometric pressure sensor is NOT detected.
INFO: TTGO T-Beam rev. 08 is detected.
Flash memory ID: 684016
INFO: GNSS module HW version: 00040007
INFO: GNSS module FW version: 7.03 (45969)
Hostname: SoftRF-24821c
Wait for WiFi connection.
....................
Can not connect to WiFi station. Go into AP mode.
Setting soft-AP configuration ... Ready
Setting soft-AP ... Ready
IP address: 192.168.1.1
UDP server has started at port: 12389
HTTP server has started at port: 80
...
$GPRMC,132506.00,A,1234.93358,N,01234.80473,E,0.121,,040820,,,A*7F
$GPGGA,132506.00,1234.93358,N,01234.80473,E,1,09,1.07,0.9,M,45.7,M,,*5F
$GPGSA,A,3,11,01,22,32,08,03,28,21,27,,,,1.70,1.07,1.32*05
$PFLAU,0,1,2,1,0,,0,,,*4C
$GPRMC,132507.00,A,1234.93362,N,01234.80473,E,0.217,,040820,,,A*71
$GPGGA,132507.00,1234.93362,N,01234.80473,E,1,09,1.07,0.8,M,45.7,M,,*56
$GPGSA,A,3,11,01,22,32,08,03,28,21,27,,,,1.70,1.07,1.32*05
$PFLAA,3,0,0,0,2,24821C!FLR_24821C,0,,0,0.0,8*77
$PFLAU,1,1,2,1,3,-122,2,0,0,24821C*2E
$GPRMC,132508.00,A,1234.93369,N,01234.80472,E,0.267,,040820,,,A*73
$GPGGA,132508.00,1234.93369,N,01234.80472,E,1,09,1.07,0.8,M,45.7,M,,*53
$GPGSA,A,3,11,01,22,32,08,03,28,21,27,,,,1.70,1.07,1.32*05
$PFLAA,3,0,0,0,2,24821C!FLR_24821C,0,,0,0.0,8*77
$PFLAU,1,1,2,1,3,-122,2,0,0,24821C*2E
$GPRMC,132509.00,A,1234.93379,N,01234.80481,E,0.504,,040820,,,A*7D
$GPGGA,132509.00,1234.93379,N,01234.80481,E,1,09,1.07,0.7,M,45.7,M,,*50
$GPGSA,A,3,11,01,22,32,08,03,28,21,27,,,,1.70,1.07,1.32*05
$PFLAA,3,0,0,0,2,24821C!FLR_24821C,0,,0,0.0,8*77
...

## To Reproduce

For me - just use the device outside or in the aircraft.

## Expected behavior

The device ID should be filtered out just like in closed issue #76 were the handling was fixed in Protocol_OGNTP.cpp

JEK58, 2020-08-04:
I had the same problem a few times but could not reproduce it.

lyusupov, 2020-08-04:
Report from anonymous user.

**Resolution**: invalid

closed 2020-08-04 by lyusupov

BeenThereScrewedUp, 2020-08-04:
You can close it - but the error persists. Too bad for all the other users.  I will fix it myself.