lyusupov/SoftRF #91: Device ID (only 5 chars), too short for OGN registration

FlyingBird22, 2020-08-15:
## Hardware

- [ ]   Standalone
- [x]   Prime Mark II
- [ ]   Dongle
- [ ]   SkyView EZ
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   UAT978 module
- [ ]   Flight Recorder

[image]

## Firmware version

[image]

## Firmware settings

[image]

## Describe the bug (or ask a question)

Device Id has a string length of only 5 alpha numeric characters: "CEF34"
It is not possible to register an Id with only 5 characters at OGN, there is says: "Device ID format incorrect"

## To Reproduce

Don't know how to change this.

## Expected behavior

Device Id should have 6 alpha numeric characters.

mreinart, 2020-08-15:
Q: Which ID is seen by other devices or OGN ground station receiving packets from yours? 
Q: you have trace output from console?

FlyingBird22, 2020-08-15:
Connected to XCSoar on smartphone:

[image]

lyusupov, 2020-08-15:
Report from anonymous user.
**Resolution**: invalid

closed 2020-08-15 by lyusupov

CazYokoyama, 2020-08-18:
It seems FlyingBird22 has a problem when he register his device on OGN Devices database, https://ddb.glidernet.org/. The device Id on SoftRF Status is reported by handleRoot() in WebHelper.cpp as 
  <tr><th align=left>Device Id</th><td align=right>%X</td></tr>\
 ThisAircraft.addr

Because of %X, the leading 0 is suppressed. For  OGN Devices database, just add 0, i.e. 0CEF34 as a temporal solution. As a  permanent solution,
"%X" should be "%06X".
I will create a pull request.

FlyingBird22, 2020-08-19:
The good news: at OGN the number was transferred with leading 0, e.g. 0CEF34
To get better html display, at _WebHelper.cpp_ we should replace **%X** by **%.6X** in _line 597_