lyusupov/SoftRF #99: UAV version broken after RC2

Hasi123, 2020-09-25:
## Hardware

- [ ]   Standalone
- [ ]   Prime Mark II
- [ ]   Uni
- [ ]   Dongle
- [ ]   SkyView EZ
- [ ]   Raspberry Edition
- [X]   UAV
- [ ]   UAT978 module
- [ ]   Flight Recorder

[image]

## Firmware version

[image]

## Firmware settings

[image]

## Describe the bug (or ask a question)
With firmware versions over RC2, the tracker stops working after selecting UAV mode in the WebUI and rebooting. After this it is not even possible to connect to wifi and access the WebUI.

## To Reproduce
Build the UAV version and upload a newer firmware than RC2. Connect to the device and select UAV mode in the WebUI.

## Expected behavior
Not to crash

Hasi123, 2020-09-28:
Ok, it seems, that RC7 also works with my setup, only RC8 crashes, when UAV mode is selected.

Hasi123, 2020-09-29:
Update 2: RC8 works with ESP32 in UAV mode. The problem only exists with ESPESP8266.

lyusupov, 2021-03-12:
**Resolution:** fixed in https://github.com/lyusupov/SoftRF/commit/2e3ac3a308777571ced53687769faea292fe40c0

closed 2021-03-12 by lyusupov