lyusupov/SoftRF #101: Сross-platform smartphone/tablet app

dmitrydudkin, 2020-11-11:
## Hardware

- [ ]   Standalone
- [x]   Prime Mark II
- [ ]   Uni
- [ ]   Dongle
- [ ]   SkyView EZ
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   UAT978 module
- [ ]   Flight Recorder

Здравствуйте,  @lyusupov!

Давно слежу за проектом и даже иногда тестирую в небе Кисловодска) Увидел в [CONTRIBUTING.md](https://github.com/lyusupov/SoftRF/blob/bde32340a27f8b412649a77573aa9da6b7a74b5e/)
то что смогу реализовать, а именно:
> create and maintain cross-platform (Flutter) smartphone/tablet app to monitor status and change settings of the following SoftRF Editions

Правда специализируюсь не на Flutter, а на React Native. 
Сделал простой прототип приложения, как альтернативу текущему web-интерфейсу. Выглядит примерно так:

[image] 

[image]

В саму прошивку пока сильно не лез, но есть задумка сразу сделать так чтобы можно было интерфейс приложения сделать динамичным, без необходимости обновления самого приложения. На устройстве достаточно будет сгенерировать json со всеми возможными настройками + текущее состояние и повесить хендлер на url типа /settings/json, а приложение уже отрисует нужный интерфейс. Думаю идея понятна.

Насколько критично использование Flutter и есть ли смысл продолжать?

lyusupov, 2020-11-12:
Выглядит вполне прилично.

Flutter не принципиален. Обновил CONTRIBUTING.

UI для моделей с Wi-Fi не предмет для инвестиций времени на это в ближайшем будущем.
Утверждение об отдельном приложении сфокусировано не моделях, в которых нет встроенного Wi-Fi.

dmitrydudkin, 2020-11-12:
Да, невнимательно прочитал CONTRIBUTING... А как предполагается связь с подобными устройствами? Bluetooth/USB OTG?

lyusupov, 2020-11-13:
## Физический/канальный уровень

Андроид:
- USB Serial
- Bluetooth SPP
- 'BLE UART'

iOS:
- 'BLE UART'

Примеры использования: [1](https://github.com/lyusupov/SoftRF/wiki/Dongle-settings) , [2](https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal) , [3](https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal) , [4](https://github.com/kai-morich/SimpleUsbTerminal) , [5](https://github.com/kai-morich/SimpleBluetoothTerminal) , [6](https://github.com/kai-morich/SimpleBluetoothLeTerminal) 

## Сообщения

NMEA

Примеры использования: [1](https://github.com/lyusupov/SoftRF/wiki/Settings) , [2](https://github.com/lyusupov/SoftRF/blob/master/software/app/interface/SoftRF.h) , [3](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SkyWatch/NMEAHelper.cpp#L467)

lyusupov, 2020-12-04:
Ticket is closed due to no further activity of the enhancement submitter.

**Resolution:** deferred

dmitrydudkin, 2020-12-07:
@lyusupov Процесс разработки оказался дольше из-за сложностей с BLE при работе с React Native.

Вот первая тестовая [сборка](https://yadi.sk/d/nMEN10D4-dhXgQ) для Android. (на iOS также работает, но сборку не делал) Тестировал на двух устройствах Prime Mark II (других нет) через BLE и USB.  Для тестирования работы BLE включал обработку в [коде](https://github.com/lyusupov/SoftRF/blob/520ea44eb8c254f7a1cf30c05a907b6ffbf7f4fb/software/firmware/source/SoftRF/src/driver/GNSS.cpp#L1075) прошивки. 

Минимальная версия Android - 4.4, тестировал на Android 5.0.2 и Android 10.

lyusupov, 2020-12-07:
Хорошо!

> Тестировал на двух устройствах Prime Mark II (других нет) через BLE и USB. Для тестирования работы BLE включал обработку в коде прошивки.

Всё правильно. Prime2 тоже можно управлять по проводам и BT, но по умолчанию неактивно из-за ненадобности ( через Wi-Fi удобнее ).
Для того, чтобы функцию активировать, достаточно раскомментировать [это](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/source/SoftRF/src/platform/ESP32.h#L229)

Есть Андроид 5.0
Сегодня опробую.

dmitrydudkin, 2020-12-07:
Внешний вид:
[image]
[image]
[image]
[image]

> Для того, чтобы функцию активировать, достаточно раскомментировать это

Да, я включал USE_NMEA_CFG. Но без правки [в блоке условий](https://github.com/lyusupov/SoftRF/blob/520ea44eb8c254f7a1cf30c05a907b6ffbf7f4fb/software/firmware/source/SoftRF/src/driver/GNSS.cpp#L1070) не заработало. Чтение данных с BLE идет только `if !defined(USE_NMEA_CFG)` (баг?)

lyusupov, 2020-12-07:
> Да, я включал USE_NMEA_CFG. Но без правки в блоке условий не заработало. Чтение данных с BLE идет только if !defined(USE_NMEA_CFG) (баг?)

Хм. Возможно. При случае проверю. Спасибо!

lyusupov, 2020-12-08:
Скиньте пустое письмо на мой адрес, указанный в firmware на странице About.
Я отвечу на него с первыми результатами опробирования приложения.

closed 2021-03-01 by lyusupov

dkm, 2025-02-13:
Hello @lyusupov and @dmitrydudkin 👋 I am looking for the corresponding source for the android application. Is it available somewhere?