lyusupov/SoftRF #106: SoftRF Dongle Edition conflict with OGN and FLARM // change of HEX-ID as prefix 0x5B is not processed

chickenhouseBjorn, 2021-01-10:
**ATTENTION** ! 

- Use this form to report bugs in released SoftRF firmware, hardware and web pages only ;
- Issue reports from anonymous users will be most likely invalidated. Reasons for that are well explained [here](https://berthub.eu/articles/posts/anonymous-help/) ;
- Prior to asking a question, look into ["Closed issues"](https://github.com/lyusupov/SoftRF/issues?q=is%3Aissue+is%3Aclosed) area to see if it was answered before ;
- No any "ideas" or new feature requests, please ! New feature requests do not subject for taking into consideration unless they come from co-developers ;
- "Enhancements" of existing features are allowed only when you are ready to implement them by yourself ;

Recommended:
- screenshot or photo width - from 600 to 800 pixels ;
- aspect ratio - 4:3, 3:4, 16:9 or 9:16.
- substitute 'space' by 'x' character to mark a hardware checkbox

--------------------

## Hardware

- [ ]   Standalone
- [ ]   Prime Mark II
- [ ]   Uni
- [X]   Dongle
- [ ]   Mini
- [ ]   SkyView EZ
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   UAT978 module
- [ ]   Flight Recorder

< photo of your hardware > (**required**)

## Firmware version
SoftRF-firmware-v1.0-rc8-S76G.dfu

<  screenshot of WebUI status page > (**required**)

## Firmware settings

<  screenshot of WebUI settings page > (**required**)

## Describe the bug (or ask a question)

I do have a SoftRF Dongle Edition flashed to current firmware, which is not visible in neither OGN nor FLARM environments, so invisible for detection units like Stratux etceteras.

As it turns out, the Dongle uses a HEX-ID for transmission starting 5B and is not processed by OGN and/or FLARM. Research and help of our fellow friends from Stratux and OGN programming revealed prefix 0x5B HEX-IDs belong to a different encoding family and were used for a service called D.S.X, so were subject to exclusion rules in receiving and processing OGN and FLARM.

SoftRF Dongle Edition therefore needs a possibility to change its HEX-ID to be something else than starting 0x5B.

## To Reproduce

just take a Dongle transmitting on a 0x5B prefix.

## Expected behavior

lyusupov, 2021-01-16:
- anonymous user and
- incomplete issue report and
- works fine to me :

[image]

[image]

<br>
<br>

**Resolution**: invalid

closed 2021-01-16 by lyusupov

lyusupov, 2021-07-19:
Reproduced with OGN software 0.2.8 
FLARM traffic is the only affected.

Every of 0x5B**** addresses have been remapped in: https://github.com/lyusupov/SoftRF/commit/f63aa9644d3af887591f346333e417c97ffed95b