lyusupov/SoftRF #116: Docs update for Dongle

hyperknot, 2021-04-17:
I couldn't update the Wiki, so I'll do it here.

1. For the AcSiP S7xG pages, the DFU mode will not be seen by any versions of Windows automatically. You'll have to go do device manager, select the device with the error icon and manually click update driver.

[image]

Navigate to this folder where you'll need to select the matching OS:
```
C:\Program Files (x86)\STMicroelectronics\Software\DfuSe v3.0.6\Bin
```

2. It's not stated when should a factory reset be performed. I guess it should be performed **after** a FW upgrade and it is recommended to do so, isn't it?

3. It should be stated that to get the boot message in Putty or screen, you have find the right timing. On my device it's 3 seconds after plugging it in. Before and it's not shown, after and it's too late.

4. Just out of interest, what does the `v1.6.5-g9-option.dfu` file do? Do we need to use it at any times? Is it a problem if someone flashes it?

lyusupov, 2021-04-19:
* report is filed not in compliance with the [template](https://github.com/lyusupov/SoftRF/blob/master/.github/ISSUE_TEMPLATE/softrf-issue-template.md) ;

**Resolution**: invalid.

closed 2021-04-19 by lyusupov

hyperknot, 2021-04-19:
Really? I'm offering my help to contribute to the Wiki, making screenshots, checking specific paths and all I get is a:
- "closed"
- "invalid"
- "report is filed not in compliance with the template"?

The wiki is closed and the template is hardware specific, with no part about fixing documentation.