lyusupov/SoftRF #121: LILYGO T5 4.7 Support

tiaanv, 2021-09-27:
Apologies if this is NOT the right place to log issues (feature request).

Looking at the code, support for the T5 (not S), is partially included in the code, yet it does not make sense.   I'm not even sure the library in use supports it.

I've tried changing code a bit and tried a few tricks, but to no avail, seems a different library will be required... Might tackle it, if you have not made it work yourself yet.  

Are you considering implementing support for it, or was it abandoned? I appreciate the fact that it might not really be a common use case, I just coincidentally have one..

lyusupov, 2021-09-27:
* report is filed not in compliance with the [template](https://github.com/lyusupov/SoftRF/blob/master/.github/ISSUE_TEMPLATE/softrf-issue-template.md) ;
* Use this form to report **bugs** in **released** SoftRF firmware, hardware and web pages only ;
* No any "ideas" or new feature requests, please ! New feature requests do not subject for taking into consideration unless they come from co-developers ;
* we do not share our plans or intentions with third-parties.

**Resolution**: invalid

closed 2021-09-27 by lyusupov

tiaanv, 2021-09-27:
## Hardware

- [ ]   Standalone
- [ ]   Badge
- [ ]   Prime Mark II
- [ ]   Dongle
- [ ]   Bracelet
- [ ]   Uni
- [ ]   Mini
- [X]   SkyView EZ
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   UAT978 module
- [ ]   Flight Recorder

[image]
[image]

## Firmware version

0.17
[image]

## Firmware settings

[image]

## Describe the bug

T5 4.7 Display not working.
Log from serial output:

```
SkyView-ESP32 FW.REV: 0.12 DEV.ID: 7e574d34
Copyright (C) 2019-2021 Linar Yusupov. All rights reserved.
EEPROM version: 28

Intializing E-ink display module (may take up to 10 seconds)... failed!
Hostname: SkyView-574d34
Setting soft-AP configuration ... Ready
Setting soft-AP ... Ready
IP address: 192.168.1.1
HTTP server has started at port: 80
```

## To Reproduce

1. Install Firmware
2. Configure via webUI (Select T5 4.7)

## Expected behavior

Screen to work.