lyusupov/SoftRF #125: wrong chip id 0xafd8

kolod1am, 2021-11-22:
## Hardware

- [ ]   Standalone
- [ ]   Badge
- [x]   Prime Mark II
- [ ]   Dongle
- [ ]   Bracelet
- [ ]   Uni
- [ ]   Mini
- [ ]   SkyView EZ
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   UAT978 module
- [ ]   Flight Recorder

[image]

## Firmware version

SoftRF-firmware-v1.0.bin

## Firmware settings

Does not apply.

## Describe the bug

After writing SoftRF to the board, no WiFi SSID appears, no bluetooth device can be scanned. No wonder, as reading from ttyUSB0 gains:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
wrong chip id 0xafd8
wrong chip id 0xafd8
wrong chip id 0xafd8
wrong chip id 0xafd8
wrong chip id 0xafd8
wrong chip id 0xafd8
wrong chip id 0xafd8
wrong ch�ets Jul 29 2019 12:21:46
```
and running.

The board came with Meshtastic. It boots and the bluetooth device appears in smartphone. So I know that the board is operational. I wonder where message "wrong chip id" comes from as I cannot find it in SoftRF source code nor in the binary.

## To Reproduce
~/.local/bin/esptool.py --chip esp32 -p /dev/ttyUSB0  write_flash 0x00000 ESP32/SoftRF-firmware-v1.0.bin

## Expected behavior
SoftRF boots to operational state.

lyusupov, 2021-11-22:
Reporter is trying to flash **ESP32** SoC with firmware binary been built for **ESP8266** SoC.
* [ESP32 binaries location](https://github.com/lyusupov/SoftRF/tree/master/software/firmware/binaries/ESP32/SoftRF)
* [ESP32 firmware installation instructions](https://github.com/lyusupov/SoftRF/blob/master/software/firmware/binaries/README.md#esp32)

**Resolution:** invalid

closed 2021-11-22 by lyusupov

kolod1am, 2021-11-22:
Many thanks for your guidance. I have got ESP32 binary from [ESP32 binaries location](https://github.com/lyusupov/SoftRF/tree/master/software/firmware/binaries/ESP32/SoftRF), unzipped the archive and did separate esptool write_flash command for each line in firmware.txt. In my case it was:
```
~/.local/bin/esptool.py --chip esp32 -p /dev/ttyUSB0  write_flash 0xe000  boot_app0.bin
~/.local/bin/esptool.py --chip esp32 -p /dev/ttyUSB0  write_flash 0x1000  bootloader_dio_80m.bin
~/.local/bin/esptool.py --chip esp32 -p /dev/ttyUSB0  write_flash 0x8000  SoftRF.ino.partitions.bin
~/.local/bin/esptool.py --chip esp32 -p /dev/ttyUSB0  write_flash 0x10000 SoftRF.ino.bin
```

Now SoftRF is up and running. I can see its WiFi ESSID.