lyusupov/SoftRF #126: Very fast battery drain when powered off - LilyGO® T-Echo Badge Version

b8fly, 2021-11-24:
**ATTENTION** ! 

- Use this form to report bugs in released SoftRF firmware, hardware and web pages only ;
- Issue reports from anonymous users will be most likely invalidated. Reasons for that are well explained [here](https://berthub.eu/articles/posts/anonymous-help/) ;
- Prior to asking a question, look into ["Closed issues"](https://github.com/lyusupov/SoftRF/issues?q=is%3Aissue+is%3Aclosed) area to see if it was answered before ;
- No any "ideas" or new feature requests, please ! New feature requests do not subject for taking into consideration unless they come from co-developers ;
- "Enhancements" of existing features are allowed only when you are ready to implement them by yourself ;

Recommended:
- screenshot or photo width - from 600 to 800 pixels ;
- aspect ratio - 4:3, 3:4, 16:9 or 9:16.
- substitute 'space' by 'x' character to mark a hardware checkbox

--------------------

## Hardware

- [ ]   Standalone
- [x]   Badge
- [ ]   Prime Mark II
- [ ]   Dongle
- [ ]   Bracelet
- [ ]   Uni
- [ ]   Mini
- [ ]   SkyView EZ
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   UAT978 module
- [ ]   Flight Recorder

[image]

## Firmware version

[image]

## Firmware settings

[image]
[image]

## Describe the bug
When powered off (Normal Off Screen) the T-Echo completely drains battery very fast (in days).
(Is this normal and expected behavior? If I do not use the fully charged device for a few days the battery is drained totally and I have first to recharge it before use.)

## To Reproduce
Fully charge the device. Shut off and wait a few days.

## Expected behavior
With a sleep current as documented of 0.25mA the expected discharging time of the powered off device should be around 140 days (850mAh battery). So the drawn current of the device seems to be much higher (maybe around 5mA did not measure it precisely).

closed 2021-11-24 by lyusupov

lyusupov, 2021-11-28:
Inspect, if this resistor is soldered on your particular T-Echo board.

[image]

b8fly, 2021-11-28:
Thanks for coming back to me. On my T-Echo board there seems to be no such resistor..
[image]
.

lyusupov, 2021-11-28:
Your board has Rw missing but R37 is installed.

[image]

You can try to fix your issue with soldering a 'bridge' over Rw pads.

b8fly, 2021-11-28:
Thank you for your research.
A normal bridge (simple wire - no resistor) will solve the problem? Did I understand that correctly? I don't know if I can do this without destroying the board. 
I will probably simply put a switch between the battery and the board.
Thanks again for solving the problem an keep up the great work!

lyusupov, 2021-12-05:
T-Echo with Zetta Device ZD25WQ16B flash memory chip are the only affected.

**Resolution:** fixed

T-Echo firmware binaries were updated in this commit: https://github.com/lyusupov/SoftRF/commit/97fe84ade91bdd47c4ef62efd31fb4731fe61832