lyusupov/SoftRF #135: T-Beam Turns On When Connecting Charger

micheal813, 2022-12-16:
I'm powering the T-Beam with a 18650 lipo. I turn off the device and connect a charger via usb. When I do, the device powers back on.

Can you make it so that if the power is off, and a battery is inserted, that it does not turn back on when connecting power via usb to charge?

Thank you.

cdeerinck, 2022-12-16:
After you connect to USB, just hold down the power button to turn the software off.  I have no idea if the charging will continue to occur at that point.  If it does, you have your solution.  If it does not, then the controller is being used to assist in the recharge, so your answer would be that it can't happen.  It would be nice for you to test that, and report back here.

micheal813, 2022-12-16:
Yes, if I hold the power button, it will turn back off. It also continues charging. I just thought it would be more convenient if you did not have to turn it back off when connecting a charger.

Thank you.

lewisxhe, 2022-12-17:
This is set by the PMU. By default, when connecting USB, it will automatically power on by default. It cannot be changed

micheal813, 2022-12-17:
Ok. I didn't know that. Thank you for responding.

closed 2022-12-17 by micheal813

lyusupov, 2022-12-17:
* report is filed not in compliance with the [template](https://github.com/lyusupov/SoftRF/blob/master/.github/ISSUE_TEMPLATE/softrf-issue-template.md) ;
* anonymous user

**Resolution**: invalid