lyusupov/SoftRF #139: Skyview EPD text oclock problem

nbonniere, 2023-02-18:
**ATTENTION** ! 

- Use this form to report bugs in an [**Official Release**](https://github.com/lyusupov/SoftRF/releases) of SoftRF firmware, hardware and web pages only ;
- Issue reports from anonymous users will be most likely invalidated. Reasons for that are well explained [here](https://berthub.eu/articles/posts/anonymous-help/) ;
- Prior to asking a question, look into ["Closed issues"](https://github.com/lyusupov/SoftRF/issues?q=is%3Aissue+is%3Aclosed) area to see if it was answered before ;
- No any "ideas" or new feature requests, please ! New feature requests do not subject for taking into consideration unless they come from co-developers ;
- "Enhancements" of existing features are only allowed when you are ready to implement them by yourself. See [CONTRIBUTING](https://github.com/lyusupov/SoftRF/blob/master/CONTRIBUTING.md) list for details ;

Recommended:
- screenshot or photo width - from 600 to 800 pixels ;
- aspect ratio - 4:3, 3:4, 16:9 or 9:16.
- substitute 'space' by 'x' character to mark a hardware checkbox

--------------------

## Hardware

- [ ]   Standalone
- [ ]   Badge
- [ ]   Prime Mark II
- [ ]   Dongle
- [X]   SkyView EZ
- [ ]   Uni
- [ ]   Mini
- [ ]   Bracelet
- [ ]   Academy
- [ ]   Lego
- [ ]   Balkan
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   WebTop Serial
- [ ]   UAT978 module
- [ ]   Flight Recorder

< photo of your hardware > (**required**)

[image]

## Firmware version

<  screenshot of WebUI and/or [**SoftRF Tool**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool) status page > (**required**)
[image]

## Firmware settings

<  screenshot of WebUI and/or [**SoftRF Tool**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool) settings page > (**required**)
[image]

## Describe the bug
oclock value incorrect on EPD Text display
- file View_Text_EPD.cpp
- missing brackets around atan2f function - converted to int before *180/PI
    bearing = (int) ( atan2f(traffic[EPD_current - 1].fop->RelativeNorth,
                           traffic[EPD_current - 1].fop->RelativeEast) * 180.0 / PI );  /* -180 ... 180 */

## To Reproduce
Set up the conditions for a target near 2 oclock, and it will indicate 1 oclock instead.

## Expected behavior
At 2 oclock it should show 2 oclock

lyusupov, 2023-02-23:
Fixed is https://github.com/lyusupov/SoftRF/commit/f8f38b28b0ae8fd975b014a8ceb02befc23c8f87

closed 2023-02-23 by lyusupov

nbonniere, 2023-02-24:
Note for Skyview: I found that the file 'feet.wav' is in voice1, but is missing in voice2 and voice3.