lyusupov/SoftRF #140: Wrong WiFi Password Prime Edition MkII

jhonny001, 2023-02-27:
## Hardware

- [ ]   Standalone
- [ ]   Badge
- [X]   Prime Mark II
- [ ]   Dongle
- [ ]   SkyView EZ
- [ ]   Uni
- [ ]   Mini
- [ ]   Bracelet
- [ ]   Academy
- [ ]   Lego
- [ ]   Balkan
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   WebTop Serial
- [ ]   UAT978 module
- [ ]   Flight Recorder

[image]

## Firmware version

[SoftRF-firmware-v1.2-ESP32] (I can't log in to show a SoftRF screenshot neither the LCD is working)

## Firmware settings

Stock settings. This is a fresh install on a new board. I can't provide a screenshot as I have no access to WiFi or BT.

## Describe the bug
The WiFi password is reported to be incorrect. (Password "12345678"). I tried from different devices (2 Android and one Windows laptop) the three of them reported password is incorrect.

## To Reproduce
Fresh install in LILYGO board TTGO T-BEAM V11 with SX1262 and M8N GPS module. FW SoftRF-firmware-v1.2-ESP32.  Try to connect using WiFi. 

## Expected behavior
Connect to the WiFi hotspot to configure.

lyusupov, 2023-02-27:
* [Disclaimer](https://github.com/lyusupov/SoftRF/wiki/Prime-Edition-MkII#disclaimer)

**Resolution**: unable to reproduce

lyusupov, 2023-02-27:
https://groups.google.com/g/openglidernetwork/c/G2kAqVuqpKg

[image]

[image]

closed 2023-02-27 by lyusupov

jhonny001, 2023-02-27:
Hi Linar, thank you very much for the support. Could you please grant me access to the Google Groups thread so I can check how Doug fixed the issue? Cheers,