lyusupov/SoftRF #145: LilyGo  no IMU after update

Jeripimo, 2023-06-27:
**ATTENTION** ! 

- Use this form to report bugs in an [**Official Release**](https://github.com/lyusupov/SoftRF/releases) of SoftRF firmware, hardware and web pages only ;
- Issue reports from anonymous users will be most likely invalidated. Reasons for that are well explained [here](https://berthub.eu/articles/posts/anonymous-help/) ;
- Prior to asking a question, look into ["Closed issues"](https://github.com/lyusupov/SoftRF/issues?q=is%3Aissue+is%3Aclosed) area to see if it was answered before ;
- No any "ideas" or new feature requests, please ! New feature requests do not subject for taking into consideration unless they come from co-developers ;
- "Enhancements" of existing features are only allowed when you are ready to implement them by yourself. See [CONTRIBUTING](https://github.com/lyusupov/SoftRF/blob/master/CONTRIBUTING.md) list for details ;

Recommended:
- screenshot or photo width - from 600 to 800 pixels ;
- aspect ratio - 4:3, 3:4, 16:9 or 9:16.
- substitute 'space' by 'x' character to mark a hardware checkbox

--------------------

## Hardware

- [ ]   Standalone
- [ ]   Badge
- [ ]   Prime Mark II
- [ ]   Dongle
- [ ]   SkyView EZ
- [ ]   Uni
- [ ]   Mini

- [ ]   Academy
- [ ]   Lego
- [ ]   ES Edition
- [ ]   Balkan
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   WebTop Serial
- [ ]   WebTop USB
- [ ]   UAT978 module
- [ ]   Flight Recorder

< photo of your hardware > (**required**)

## Firmware version 1.3

<  screenshot of WebUI and/or [**SoftRF Tool**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool) status page > (**required**)

## Firmware settings

<  screenshot of WebUI and/or [**SoftRF Tool**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool) settings page > (**required**)

## After update from firmware v1.1 main screen says IMU N/A

## Update firmware from 1.1 to 1.3

## working IMU

Jeripimo, 2023-06-27:
It's the lilygo t-echo

lyusupov, 2023-06-28:
* report is filed not in compliance with the [template](https://github.com/lyusupov/SoftRF/blob/master/.github/ISSUE_TEMPLATE/softrf-issue-template.md) ;
* anonymous user

**Resolution**: invalid

closed 2023-06-28 by lyusupov