lyusupov/SoftRF #152: Standalone version problem

legacy60000, 2024-01-26:
**ATTENTION** ! 

- Use this form to report bugs in an [**Official Release**](https://github.com/lyusupov/SoftRF/releases) of SoftRF firmware, hardware and web pages only ;
- Issue reports from anonymous users will be most likely invalidated. Reasons for that are well explained [here](https://berthub.eu/articles/posts/anonymous-help/) ;
- Prior to asking a question, look into ["Closed issues"](https://github.com/lyusupov/SoftRF/issues?q=is%3Aissue+is%3Aclosed) area to see if it was answered before ;
- No any "ideas" or new feature requests, please ! New feature requests do not subject for taking into consideration unless they come from co-developers ;
- "Enhancements" of existing features are only allowed when you are ready to implement them by yourself. See [CONTRIBUTING](https://github.com/lyusupov/SoftRF/blob/master/CONTRIBUTING.md) list for details ;

Recommended:
- screenshot or photo width - from 600 to 800 pixels ;
- aspect ratio - 4:3, 3:4, 16:9 or 9:16.
- substitute 'space' by 'x' character to mark a hardware checkbox

--------------------

## Hardware

- [x]   Standalone
- [ ]   Badge
- [ ]   Prime Mark III
- [ ]   Prime Mark II
- [ ]   Dongle
- [ ]   SkyView EZ
- [ ]   SkyView Pico
- [ ]   Uni
- [ ]   Ham
- [ ]   Midi
- [ ]   Mini

- [ ]   Academy
- [ ]   Lego
- [ ]   ES Edition
- [ ]   Balkan
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   WebTop Serial
- [ ]   WebTop USB
- [ ]   UAT978 module
- [ ]   Flight Recorder

< photo of your hardware > (**required**)
[image]
[image]
[image]
[image]
## Firmware version

<  screenshot of WebUI and/or [**SoftRF Tool**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool) status page > (**required**)

## Firmware settings

<  screenshot of WebUI and/or [**SoftRF Tool**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool) settings page > (**required**)

## Describe the bug
After startup (changing colors etc..) the LED ring stays in kinda white dimmed state and buzzer stays continuesly on. After gps lock it goes to boot loop. No effect by removing the LED ring and led bar or anyway varying with HW pieces.

Checks done:
- Erased NODEMCU before flashing FW
- Used several different HW pieces
- Power checked twice 3,30V
- Flashed different versions of FW V1.0,V1.1...1.4

## To Reproduce

## Expected behavior

legacy60000, 2024-01-26:
Additional tryouts:

Removed the node mcu from shield:

Suplied 5v for LED ring - nil change
Tryed different manufacturer of LED ring - nil change
Went through all versions of FW - nil change

lyusupov, 2024-02-11:
Works fine to me.

<hr>

[image]

<hr>

<br>
<br>

**Resolution**: unable to reproduce

<br>

closed 2024-02-11 by lyusupov