lyusupov/SoftRF #154: Issues with Flashing binaries to SkyView EZ

BrianJSavage, 2024-02-10:
The board I ordered of AliExpress is:
LILYGO® TTGO T5S V2.8 ESP32 2.7 Inch Screen E-Paper Module I2S DAC MAX98357A Microphone ICS43434 SD Card Buletooth Board

I downloaded the Flash download tool from Expressif.
I chose ESP32 on startup, and copied in the binaries as shown in the screenshot below.
After running screen is now locked on one of the initial startup screens.
The app is different to your sample in the repo, in that there is no flash size panel.
Also, there were two options for the SkyView.ino.bin file on the repo,  SkyView_C105.ino.bin and C209, I tried both.
The log from the Com window is also below.
There are some errors in the log, and I am unsure as to what to try next.
Help!
Cheers
Brian

[image]

Log:

F:\My Documents D\Downloads\flash_download_tool_3.9.5\flash_download_tool_3.9.5
F:\My Documents D\Downloads\SoftRF-master\SoftRF-master\software\firmware\binaries\ESP32\SkyView\SkyView-firmware-v0.13-ESP32\boot_app0.bin
1 kill
F:\My Documents D\Downloads\SoftRF-master\SoftRF-master\software\firmware\binaries\ESP32\SkyView\SkyView-firmware-v0.13-ESP32
F:\My Documents D\Downloads\SoftRF-master\SoftRF-master\software\firmware\binaries\ESP32\SkyView\SkyView-firmware-v0.13-ESP32\bootloader_dio_80m.bin
2 kill
F:\My Documents D\Downloads\SoftRF-master\SoftRF-master\software\firmware\binaries\ESP32\SkyView\SkyView-firmware-v0.13-ESP32
F:\My Documents D\Downloads\SoftRF-master\SoftRF-master\software\firmware\binaries\ESP32\SkyView\SkyView-firmware-v0.13-ESP32\SkyView.ino.partitions.bin
3 kill
F:\My Documents D\Downloads\SoftRF-master\SoftRF-master\software\firmware\binaries\ESP32\SkyView\SkyView-firmware-v0.13-ESP32
F:\My Documents D\Downloads\SoftRF-master\SoftRF-master\software\firmware\binaries\ESP32\SkyView\SkyView-firmware-v0.13-ESP32\SkyView_C105.ino.bin
4 kill
.
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
NO XMC flash  detected!
FLASH_CRYPT_CNT 0
ABS_DONE_0 False

is stub and send flash finish
.
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 115200
Changed.
FLASH_CRYPT_CNT 0
ABS_DONE_0 False
.
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
NO XMC flash  detected!
FLASH_CRYPT_CNT 0
ABS_DONE_0 False

is stub and send flash finish
F:\My Documents D\Downloads\SoftRF-master\SoftRF-master\software\firmware\binaries\ESP32\SkyView\SkyView-firmware-v0.13-ESP32
F:\My Documents D\Downloads\SoftRF-master\SoftRF-master\software\firmware\binaries\ESP32\SkyView\SkyView-firmware-v0.13-ESP32\SkyView_C209.ino.bin
.
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 115200
Changed.
FLASH_CRYPT_CNT 0
ABS_DONE_0 False
.
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
NO XMC flash  detected!
FLASH_CRYPT_CNT 0
ABS_DONE_0 False

is stub and send flash finish
F:\My Documents D\Downloads\SoftRF-master\SoftRF-master\software\firmware\binaries\ESP32\SkyView\SkyView-firmware-v0.13-ESP32

--------------------

## Hardware

- [ ]   SkyView EZ
[image]
[image]

## Firmware version

<  screenshot of WebUI and/or [**SoftRF Tool**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool) status page > (**required**)

## Firmware settings

<  screenshot of WebUI and/or [**SoftRF Tool**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool) settings page > (**required**)

## Describe the bug

## To Reproduce

## Expected behavior

lyusupov, 2024-02-10:
**No any of the checkmarks applied:**

[image]

One should follow the [**Quick Start**](https://github.com/lyusupov/SoftRF/wiki/SkyView.-Quick-start) instructions to succeed.

**Resolution:** invalid

closed 2024-02-10 by lyusupov

BrianJSavage, 2024-02-10:
Hi Linar,

 

Thanks for very prompt response

 

Duh, of course. Never done this before. All working now and connected to the T-Beam Supreme.

 

I have been testing using a nearby (3m away) T-Echo, along with XcSoar, all works like a dream!

 

Now to get it wired into the glider.

 

Thanks again for your help,

 

Brian

 

image.png (view on web) <https://github.com/lyusupov/SoftRF/assets/5849637/7fb40fb8-ce9c-4862-b863-bab2514bdcb5>