lyusupov/SoftRF #157: Bad aircraft data

clydebarrow, 2024-03-30:
**ATTENTION** ! 

- Use this form to report bugs in an [**Official Release**](https://github.com/lyusupov/SoftRF/releases) of SoftRF firmware, hardware and web pages only ;
- Issue reports from anonymous users will be most likely invalidated. Reasons for that are well explained [here](https://berthub.eu/articles/posts/anonymous-help/) ;
- Prior to asking a question, look into ["Closed issues"](https://github.com/lyusupov/SoftRF/issues?q=is%3Aissue+is%3Aclosed) area to see if it was answered before ;
- No any "ideas" or new feature requests, please ! New feature requests do not subject for taking into consideration unless they come from co-developers ;
- "Enhancements" of existing features are only allowed when you are ready to implement them by yourself. See [CONTRIBUTING](https://github.com/lyusupov/SoftRF/blob/master/CONTRIBUTING.md) list for details ;

Recommended:
- screenshot or photo width - from 600 to 800 pixels ;
- aspect ratio - 4:3, 3:4, 16:9 or 9:16.
- substitute 'space' by 'x' character to mark a hardware checkbox

--------------------

## Hardware

- [ ]   Standalone
- [x]   Badge
- [ ]   Prime Mark III
- [ ]   Prime Mark II
- [ ]   Dongle
- [ ]   SkyView EZ
- [ ]   SkyView Pico
- [ ]   Uni
- [ ]   Ham
- [ ]   Midi
- [ ]   Mini

- [ ]   Academy
- [ ]   Lego
- [ ]   ES Edition
- [ ]   Balkan
- [ ]   Raspberry Edition
- [ ]   UAV
- [ ]   WebTop Serial
- [ ]   WebTop USB
- [ ]   UAT978 module
- [ ]   Flight Recorder

< photo of your hardware > (**required**)
[image]

## Firmware version

1.4 installed from Releases page via UF2.

<  screenshot of WebUI and/or [**SoftRF Tool**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool) status page > (**required**)
[image]

## Firmware settings

<  screenshot of WebUI and/or [**SoftRF Tool**](https://github.com/lyusupov/SoftRF/wiki/SoftRF-Configuration-Tool) settings page > (**required**)
[image]

## Describe the bug

Until recently this was inter-operating well with Flarm devices, using the Legacy protocol. Now it receives data, but the reported aircraft data is completely bogus. The same behaviour was observed with firmware 1.3, updating did not change this.

## To Reproduce

Receive Flarm transmissions with the T-Echo SoftRF model.

## Expected behavior

Reported data will be valid.

lyusupov, 2024-03-30:
- "Enhancements" of existing features are only allowed when you are ready to implement them by yourself. See [CONTRIBUTING](https://github.com/lyusupov/SoftRF/blob/master/CONTRIBUTING.md) list for details ;

<br>
<br>

[image]

<br>
<br>

**Resolution:** invalid

closed 2024-03-30 by lyusupov

clydebarrow, 2024-03-30:
I was aware that feature requests were not welcome here, but perhaps you could explain more why this would be considered an enhancement. What information am I missing? Has there been a change in the Flarm protocol recently that renders SoftRF incompatible?

lyusupov, 2024-03-30:
https://groups.google.com/g/openglidernetwork/c/WLVx0q3PSzc

lyusupov, 2024-05-14:
Volunteer contributors are welcome: https://pastebin.com/YB1ppAbt

[image]

lyusupov, 2024-05-20:
[image]